greatscottgadgets/hackrf #683: HackRF LNA not working

sashatytko, 2019-12-31:
Hello!
Few days ago I realized that my LNA not working correctly. When I increase it just noise appeared.I think that my amp is working correctly. I have just updated firmware.
Could you say what components should be replaced?
Please see video:
[https://youtu.be/9WW35kRIiq8](url)

JonaldJohnston, 2020-01-02:
the PA and LNA both us a MGA-81563, or so i have heard. im not sure where its located on the PCB though.

I accidentally blew out my PA once and GSG was happy to replace my HackRF free of charge (except for shipping on they way down), so id email them first if you don't want to muck around with the unit itself.

sashatytko, 2020-01-16:
Not fixed yet

miek, 2020-01-16:
I took a look at the video & I agree that the LNA has failed. If you bought your HackRF from an official reseller, you can contact them to arrange a replacement.

If not and/or you want to replace it yourself, @JonaldJohnston replied above with the part number that should be replaced (MGA-81563) and they're at U13 and U25.

sashatytko, 2020-01-16:
Are you sure?Because i have heard that this component used for external amp(that is near to antenna and called RF gain in gnuradio).I have BB/IF gain not working

miek, 2020-01-16:
OK sorry, I think i understand the original question better now.

The BB/IF gain (named VGA/LNA gain in SDRSharp) behaviour looks correct to me. It's totally normal that the noise floor increases with the signal.

The TX amp looks good. The RX amp does not seem to be working correctly - you should see the signal & noise floor increase when it is enabled.

sashatytko, 2020-01-16:
I am not sure about the signal level because it is not receive anything ,even strong signal.
But on shortwave i took this photo.
On the top of waterfall  amp is on.
I try catch some strong signal and make video how turning amp (RF) infliuence on it.
[image]

sashatytko, 2020-01-16:
I have tested amp(RF) on GSM signals few minutes ago and signals increased. So it seems like my LNA(VGA) dont working correctly(almost dont work).
See video: 
https://youtu.be/IAdmBwF1mpY

miek, 2020-01-16:
Everything is working correctly in that video. The LNA/VGA gain controls are working as expected. There's not much response to the LNA gain changes initially because the VGA gain is set so low. Once  the VGA gain is set to mid-range the response to LNA gain changes are more obvious.

sashatytko, 2020-01-16:
But on rtl-sdr on the same antenna on the same frequencies there are very strong signals.
So we have RF is working correctly.
But LNA dont increase signal properly.

sashatytko, 2020-01-16:
[image]
[image]

Please see signals on 106.6 and GSM on rtl sdr

sashatytko, 2020-01-17:
I suspend that SKY13317 dont work properly.
Maybe reopen this issue?

miek, 2020-01-20:
Sure, I'll open this back up until we solve this for you.

However, I'm still not seeing a problem - you're receiving that GSM signal just fine when the gains are set to reasonable levels (24 & 32dB are good starting points for the HackRF One, in my experience).
You can't compare the two radios directly when they're quite different, and have very different bandwidth/sample rates.

sashatytko, 2020-01-21:
But earlier few weeks ago there were more signals(GSM).And now HackRF dont receive any FM station.

sashatytko, 2020-01-28:
Anyone have some idea or can help?

sashatytko, 2020-01-30:
Where is LNA(IF),VGA amplifier in HackRF on schematic?

miek, 2020-02-03:
Both of those settings control attenuators within the MAX2837.

sashatytko, 2020-02-03:
I connected wire directly from antenna to C61 and bypass U14.When i did this signals appeared!
So I make conclusion that I need replace Sky13317

sashatytko, 2020-02-03:
But when signal bypass amp?

miek, 2021-02-04:
The amp is bypassed when the amp disable (`-a 0`) flag is used with `hackrf_transfer`. gr-osmosdr and some other tools list it as 'RF gain', so that should be set to 0 for those tools.

closed 2021-02-04 by miek