greatscottgadgets/hackrf #684: hackrf dfu shows no device - 10 different cords - 3 brand new cords. Nothing

jfender1, 2019-12-31:
### Steps to reproduce
1.put hackrf in dfu mode - lights show it is in the desired mode
2.run sudo dfu-util -l
3.No devices

### Expected behaviour
listing of DFU device and access to device

### Actual behaviour
Nothing.

I think the device is toast. I purchased a portapack and it was working for some time. All of the sudden it stopped working. with the portapack installed the devices flases the rx yellow light and the portapack will not come one. Cannot program the hackrf back to normal because the usb is not working. It provides power but not data access. Changed the cord to 10 different cords as I read that could be the issue. 3 of the cords were brand new. No suck luck.

eeprommemory, 2020-01-05:
same problem portapack firmware has a bug kills usb

eeprommemory, 2020-01-07:
did you have it connected to a battery bank.

jfender1, 2020-01-07:
No. The portapack had it's own battery

eeprommemory, 2020-01-07:
that then rules out death by powerbank.
can you measure the USB port to see if the ESD U15 chip has fried

jfender1, 2020-01-07:
How do I do that?

eeprommemory, 2020-01-07:
do you have a multi meter.
if you do measure in ohms mode from antenna connector or other ground to the pins on the usb plug the end you plug in the pc.
one is ground the next one is D+ should read as meg ohms same with D- then 5V.
the D+ and D- is the ones you want to test.
if you get 0.00 ohms port is fried.

jfender1, 2020-01-07:
When I plug it in the hackrf powers on. It just wont connect to the pc. I
will try to test to usb connection today and let you know the results.
Thanks for all the help

JonaldJohnston, 2020-01-08:
same issue, usb is completely unresponsive except for power. #681

caitho, 2020-05-23:
I have gad this issue for close to a year,  My Esd chip looks intact and does not get hot but it seems likely to be the trouble as i also used PortaPack h2. But there should be a way to get it working through GPIO right ? The PortaPack can control it and display data from it so it must be theoretically possible ?!

straithe, 2021-01-28:
Are you still experiencing this issue?

caitho, 2021-01-31:
Yes, it is not registering on any device at all. It gets power but there
seems to be absolutely no data transferred. I've tried on android windows
linux and chrome os . But it works on the portal pack (though very
unstable) waterfall works 1 out of 5 tries (approximately)

straithe, 2021-02-05:
Which portapack are you using H1 or H2?

caitho, 2021-02-05:
Portapack H1
Worbel (dot) de

josmargoncal, 2021-02-08:
I have the same problem, nothing new?

straithe, 2021-02-14:
Since the original poster has not responded I will assume that their issue has been solved and I will be closing this ticket. For the rest of you, please open your own issues so I can help each of you directly.

closed 2021-02-14 by straithe

AndyH76, 2024-12-11:
Has anyone solved the USB problem? The same problem on my HackRF+Portapack H1