greatscottgadgets/hackrf #686: need bin file for U20 W25Q80BV

eeprommemory, 2020-01-05:
need bin file for U20 W25Q80BV
usb no longer works dfu mode does not work want to reprogram spi chip need file thanks.

miek, 2020-01-06:
You can find that in the release package under `firmware-bin` as `hackrf_one_usb.bin` https://github.com/mossmann/hackrf/releases/tag/v2018.01.1

miek, 2020-01-06:
Also, could you give a bit more detail about why you think a PortaPack firmware bug is causing USB issues (as you mentioned on #684 and #681)?

eeprommemory, 2020-01-06:
i did not see a burnable file in the release package.
chip size is 1mb only file i saw was 22kb.
and i think the software loads it into certain locations so it is not a full image.
it was working fine then it crashed so i reset it and all seems good then i need to use it as a SDR and nothing.
two computers several usb cords nothing.

eeprommemory, 2020-01-06:
so the hope is put original firmware back in and get usb working again and update CPLD.

miek, 2020-01-06:
That .bin file is written to address 0 on the SPI flash as-is. It's only 22kB because that's the size of the firmware - there's no need to pad it out to 1MB.

However, if DFU is not working then this is really unlikely to be a firmware problem. The DFU bootloader is burned in ROM on the LPC4320 & is selected with a dedicated pin. There's no way to overwrite or damage it with firmware updates.

eeprommemory, 2020-01-07:
connecting hackrf to a battery bank seems to be the issue not software.
checked the data lines D+ and D- D- was direct short to ground.
removed U15 LXES1T1BCC2-004 short gone but still does not work.
I normality see a 100ohm resistor on USB ports going to a 3.3V CPU R58 and R57 are 0ohm.
so CPU fried no saving it.

eeprommemory, 2020-01-07:
looked at datasheet and USB goes directly to cpu so why have R58 and R57 i have no idea.

closed 2021-04-01 by miek