greatscottgadgets/hackrf #688: duplicate samples at high sample rates

mossmann, 2020-01-08:
While RXing at sample rates of 16.5 Msps or higher I have observed incorrect sample values that are duplicates of values 16 samples in the future.

Example hex dump excerpt (captured with hackrf_transfer at 20 Msps):
```
00d68070: fd f8  ..
00d68072: fb 01  ..
00d68074: 09 fd  ..
00d68076: 02 f7  ..
00d68078: fa fd  ..
00d6807a: ff 05  ..
00d6807c: 09 02  ..
00d6807e: 07 f8  ..
00d68080: 08 03  ..
00d68082: 07 f9  ..
00d68084: ff f7  ..
00d68086: fa ff  ..
00d68088: 00 06  ..
00d6808a: 09 01  ..
00d6808c: 06 f8  ..
00d6808e: fd f9  ..
00d68090: fb 02  ..
00d68092: 04 05  ..
00d68094: 09 fd  ..
00d68096: 02 f7  ..
00d68098: fa fd  ..
00d6809a: ff 05  ..
00d6809c: 09 02  ..
00d6809e: 07 f8  ..
00d680a0: fd f8  ..
00d680a2: fb 02  ..
```
Six samples (each one byte I followed by one byte Q) in the above example starting at 00d6807a are repeated at 00d6809a.  The samples at 00d6809a are correct.  The duplicated samples at 
00d6807a are incorrect.

I've observed duplication of 2, 4, or 6 samples at a time.  When multiple duplication events occur, they typically happen 8192 (one USB buffer in firmware) samples apart.  This appears to be caused by firmware USB operations delaying execution of sgpio_isr_rx().

This GNU Radio flowgraph can be used to detect duplication events that are 8192 samples apart: https://gist.github.com/mossmann/f8999b0eafc845e95cc166d6f389f4e7

Presumably there is a similar bug on TX.

miek, 2020-01-13:
We've discussed much of this already, but just to get it written down:

I made a couple of commits to make it a little easier to see the problem here: https://github.com/miek/hackrf/tree/glitch-debug - this branch makes the CPLD output an incrementing counter (instead of RX samples), and sets the first byte of a USB bulk buffer to 0x7f (1.0).

This is a plot of samples taken at 20MSps:

[image]
[image]

I put together a proof-of-concept implementation using DMA instead of the SGPIO ISR here: https://github.com/miek/hackrf/tree/dma_wip
This seems to regularly miss a few samples on 4096 (complex) sample boundaries. We think this is due to a delay when the GPDMA is reading the next LLI in the loop. I played around with different memory layouts to position 2 (or 3) bulk buffers in different regions and place the LLIs in another region to hopefully reduce AHB contention, but that didn't help.

Next plan is to experiment with moving the SGPIO ISR to the M0.

mossmann, 2020-01-13:
Here is the M0 solution used in Rhododendron:

https://raw.githubusercontent.com/ktemkin/GreatFET/rhododendron_wip/firmware/greatfet_usb/rhododendron/rhododendron_m0.S

https://github.com/greatscottgadgets/libgreat/blob/d292362cb678edd7d6ce3342402df60ccd41c882/firmware/platform/lpc43xx/drivers/platform_reset.c#L125

Videogamer555, 2020-01-14:
If indeed it is caused by faulty firmware in the HackRF One, no driver or DLL fixes are possible. At best, on the PC side, you could detect such bad samples, and indicate to the software using it that these samples are faulty, and to ignore them. However, if you need a continuous flow of non-faulty samples (such as decoding an NTSC video signal), then you are out of luck with the current HackRF One firmware.

Your best bet is to send an email with a bug report to the official HackRF One developers mailing list which is at hackrf-dev@greatscottgadgets.com
In order to send email to them (and not have it get rejected), as well as receive the latest updates about the development of the HackRF One (such as to see when the bug you reported gets fixed), you will need to subscribe to the mailing list which you can do at https://pairlist9.pair.net/mailman/listinfo/hackrf-dev

mossmann, 2020-01-15:
> If indeed it is caused by faulty firmware in the HackRF One, no driver or DLL fixes are possible. At best, on the PC side, you could detect such bad samples, and indicate to the software using it that these samples are faulty, and to ignore them. However, if you need a continuous flow of non-faulty samples (such as decoding an NTSC video signal), then you are out of luck with the current HackRF One firmware. Alternatively, you could try to use a lower sample rate, and keep lowering it until no more glitches happened. Hopefully you will be able to lower it only slightly, and stop the glitches from happening, so that you can still receive the bandwidth required for signal decoding.
> 
> If you need to receive at 20MSPS to get the full signal, and decoding requires no lost samples, your best bet to get them to fix the firmware is to send an email with a bug report to the official HackRF One developers mailing list which is at [hackrf-dev@greatscottgadgets.com](mailto:hackrf-dev@greatscottgadgets.com)
> In order to send email to them (and not have it get rejected), as well as receive the latest updates about the development of the HackRF One (such as to see when the bug you reported gets fixed), you will need to subscribe to the mailing list which you can do at https://pairlist9.pair.net/mailman/listinfo/hackrf-dev

We are the hackrf developers.  We are discussing a firmware bug and how to fix it in firmware.  This is the place where we track bugs, though we sometimes also discuss them on the mailing list or IRC.

miek, 2020-01-15:
Just pushed up a (work-in-progress) branch that uses the M0 to collect SGPIO samples: https://github.com/miek/hackrf/tree/m0_sgpio_wip. it seems to be working well with no glitches so far!

miek, 2020-01-15:
I pushed up TX support to the https://github.com/miek/hackrf/tree/m0_sgpio_wip branch earlier and the TX signal looks good.

I've also just pushed up another branch with a test for SGPIO TX glitches: https://github.com/miek/hackrf/tree/cpld_counter_tx
This adds a counter within the CPLD which is compared with the incoming samples from SGPIO. If they don't match, pin B2AUX1 goes high and the counter is re-synchronised. It also adds an extra mode to `hackrf_transfer` to transmit a corresponding counter, which can be used with the '-G' argument.

Here's master (D0 is B2AUX1):
[image]

and here's m0_sgpio_wip:
[image]

:tada: 

I won't have much more time this week, but next week I'll prep a cleaned-up PR with the changes from m0_sgpio_wip.

I think it's probably worth getting the glitch testing stuff upstream too? I can see it being useful for testing future PRs (eg: #662) and even for end users testing USB throughput. If so, I'll tidy that up for a PR too.

mossmann, 2020-01-15:
Beautiful!  Yes, I agree that the test code would be worth keeping around.

Videogamer555, 2020-01-16:
> We are the hackrf developers. We are discussing a firmware bug and how to fix it in firmware. This is the place where we track bugs, though we sometimes also discuss them on the mailing list or IRC.

Oh ok. Sorry about that. I thought this was the webpage for hackrf_tools which I thought was 3rd party software for use with the HackRF One, but not made the developers of the actual HackRF One device and firmware. Sorry for the misunderstanding.

closed 2020-01-21 by mossmann

VioletGiraffe, 2023-10-08:
We are experiencing this issue, and the installed firmware is 2023.01.1, according to `hackrf_info`. What should we do to apply this fix? Looks exactly as described, and also has the period of 8192 samples (16 Kbytes).

martinling, 2023-10-08:
The fix discussed above was already applied, and all the relevant code has been extensively improved since. You should not be seeing this happen with 2023.01.01 firmware. If you are then the cause is probably quite different. Please open a new issue with what you're seeing, including steps to reproduce.

PR #692 (2020-01-21): Use the M0 core to handle SGPIO samples
This fixes #688