greatscottgadgets/hackrf #689: Please post compiled EXEs and DLLs

Videogamer555, 2020-01-12:
I would like a compiled copy of this. I don't have the tools needed to compile it myself.
And make sure to compile it for 32bit Windows, not 64bit Windows.

miek, 2020-01-21:
Unfortunately we can't commit to maintaining Windows binaries at this time.

If you are able to relax the requirement for 32-bit, then I recommend the 64-bit builds provided by the Pothos SDR dev enironment: https://github.com/pothosware/PothosSDR/wiki
If not, you may be able to use their scripts to build it yourself on 32-bit Windows.

closed 2020-01-21 by miek

Videogamer555, 2020-01-21:
@miek I know you don't want to commit to maintaining it. I'd just like you to compile one for me personally now. I don't care if it's kept updated to the latest version. And I'm sorry but I CANNOT relax the 32-bit requirement. IT MUST BE the 32bit version.

Right now, my HackRF One is just a $350 paperweight, sitting on my desk doing nothing, waiting for you to compile this software for me to use it.

Videogamer555, 2020-06-22:
I ended up compiling it myself. If anybody else is in the same situation as I was at this time, you can download the copy I compiled. It's the 32bit version of HackRF One Tools. You can find the download link in this post I made. https://github.com/mossmann/hackrf/issues/696