greatscottgadgets/hackrf #690: How to update HackRf one's firmware with Blinky

Gowtham-Ramesh, 2020-01-15:
It would be great if I get the supported files of blinky.c (.bin, .xsvf) to update the firmware.

miek, 2020-01-21:
Could I ask what your end goal is? The blinky project is just some test firmware and we don't maintain builds of it. It's really only useful if you're doing low-level development - in which case, you'll need to get set up to compile it anyway. There's no CPLD image (.xsvf) to go with it.

Gowtham-Ramesh, 2020-01-21:
My end goal is setting the Hackrf one to transmit in standalone mode (if
not Raspberry Pi can be attached). I tried blinky it is not working in my
computer, it pops an error (below). It would be great if you suggest me a
way to run blinky into my device as it would be a great start for
standalone programming. Also, I guess no CPLD image needed for blinky.

Building for: NMake Makefiles
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: C:/Program Files (x86)/GNU Tools ARM Embedded/5.4
2016q3/arm-none-eabi
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: C:/Program Files (x86)/GNU Tools ARM Embedded/5.4
2016q3/arm-none-eabi
-- The C compiler identification is GNU 5.4.1
-- Check for working C compiler: C:/Program Files (x86)/GNU Tools ARM
Embedded/5.4 2016q3/bin/arm-none-eabi-gcc.exe
CMake Error: Generator: execution of make failed. Make command was: nmake
/nologo cmTC_f635a\fast &&
-- Check for working C compiler: C:/Program Files (x86)/GNU Tools ARM
Embedded/5.4 2016q3/bin/arm-none-eabi-gcc.exe -- broken
CMake Error at C:/Users/Gowtham
Ramesh/cmake-3.16.2-win64-x64/cmake-3.16.2-win64-x64/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60
(message):
  The C compiler
    "C:/Program Files (x86)/GNU Tools ARM Embedded/5.4
2016q3/bin/arm-none-eabi-gcc.exe"
  is not able to compile a simple test program.
  It fails with the following output:
    Change Dir: C:/git/hackrf/firmware/blinky/CMakeFiles/CMakeTmp
    Run Build Command(s):nmake /nologo cmTC_f635a\fast && The system cannot
find the file specified
    Generator: execution of make failed. Make command was: nmake /nologo
cmTC_f635a\fast &&

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:26 (project)

-- Configuring incomplete, errors occurred!
See also "C:/git/hackrf/firmware/blinky/CMakeFiles/CMakeOutput.log".
See also "C:/git/hackrf/firmware/blinky/CMakeFiles/CMakeError.log".

miek, 2020-01-21:
This looks like the build system is failing to find `nmake` - you probably need to check your environment variables and make sure it's in the search path.

I don't really know anything about building firmware on Windows though, so I probably won't be able to help much more.

Gowtham-Ramesh, 2020-01-21:
Also I would like to know is it possible to make hackrf one to transmit
ofdm signal in standalone mode.

miek, 2020-01-21:
Yep, that should be possible.

Gowtham-Ramesh, 2020-01-23:
I transmitted OFDM signal through GNU radio. I could see
"uuuuuuuuuuuuuuuuuu" showing up in GRC cmd prompt. What's that mean ? And
what the max frequency and bandwidth, Hackrf could perform.

miek, 2020-02-03:
'u' printed out means underflow, ie: samples are not being produced fast enough for a sink block to consume them.
The HackRF can tune up to 6GHz, and the max bandwidth is 20MHz. Full specs are here: https://github.com/mossmann/hackrf/wiki/HackRF-One

closed 2020-02-12 by miek