greatscottgadgets/hackrf #696: Here's a compiled copy of the 32bit version of hackrf-tools.

Videogamer555, 2020-01-22:
I finally decided to try to compile it myself via MS Visual C++ 2015. I managed to do so after tweaking the commandline string for CMake provided on this site (which was intended to compile a 64bit version, but I was trying to make the 32bit versions, and it also contained some errors).

I've attached a ZIP file to this post, containing the compiled tools and all required dependencies. Make sure to put all the dependencies (both hackrf.dll and all the 3rd party dlls) in the same folder as the hackrf-tools EXE files, or otherwise the tools won't run. I only have them in separate folders in this ZIP package for the sake of keeping it nicely organized. Also, read the readme.txt file, as it provides more info about what I've included in this package.

[HackRF-Tools Binaries.zip](https://github.com/mossmann/hackrf/files/4096954/HackRF-Tools.Binaries.zip)

closed 2020-01-23 by mossmann

Videogamer555, 2020-01-23:
@mossmann why did you close this? Other people might also want to download the 32bit version of hackrf-tools, and they might not think to look in the closed issues to find it. If anything, my zip file with the compiled version of hackrf-tools should be added to the text on the page https://github.com/mossmann/hackrf/tree/master/host/ to make it easier for people to find.

mossmann, 2020-01-23:
I closed this issue because it asked no questions and made no requests.  It provided information that people can find by searching.