greatscottgadgets/hackrf #700: HackRF bandpass filter not working correctly on TX

Videogamer555, 2020-01-23:
I am trying to transmit a standard analog NTSC signal which has a vestigial sideband on the lower side of the carrier, and a full upper sideband. To do this, I created a signal that meets the specs for an NTSC signal that corresponds to one second's worth of signal at a sample rate of 8MHz.

The NTSC specification calls for a signal with an upper cutoff frequency of 4.2MHz above the video carrier (the full upper sideband) and a lower cutoff frequency of 0.75MHz below the video carrier (the vestigial lower sideband). I created this VSB signal using a complex FFT filter on the original unfiltered video signal.

To transmit, I used the following commandline for hackrf_transfer.exe
`hackrf_transfer.exe -f 471250000 -a 1 -x 47 -s 8000000 -b 8000000 -R -t txsignal2.data`

As you can see, I set the HackRF's own bandpass filter to the full bandwidth available at the current sample rate, by by setting -b (bandpass filter width) to the same value as -s (sample rate).

I analyzed the signal coming from the HackRF by using an AirSpy to receive the signal, and viewed its spectrum in SDR#. When centered on the carrier, everything looks ok. You can see the full upper sideband and the vestigial lower sideband. Everything below the vestigial lower sideband is blocked. I've attached a picture called "UHF TV signal centered.png" showing this.
[image]

In the next step, I looked above the signal by 2MHz, to make sure nothing was leaking above the upper 4MHz of the 8MHz HackRF passband. Nothing was (as can be seen in the attached picture "UHF TV looking above center by 2MHz.png"). So far so good.
[image]

Then I tried looking below the signal by 2MHz, to make sure nothing was leaking below the lower 4MHz of the 8MHz HackRF passband (though theoretically that would be prevented anyway, by the complex signal being sent that self-blocks all frequencies below the lower vestigial sideband). And that's when I saw something strange. Below the frequency that is 4MHz below the carrier, there suddenly is a bunch of completely unfiltered signal. This can be clearly seen in the picture "UHF TV looking below center by 2MHz.png".
[image]

It seems that either the HackRF One's hardware filter isn't working properly, or there is a glitch in the firmware that controls the filter, or there is a glitch in the software on the PC side (either hackrf.dll or hackrf_transfer.exe) that is causing this problem.

mossmann, 2020-01-24:
Firstly, HackRF One does not have a bandpass filter.  It does have a selectable baseband filter which you are interpreting to be a bandpass filter.

Secondly, your screenshots indicate that the baseband filter is working correctly.  When the HackRF is tuned to 471.25 MHz, the passband of the filter extends from 467.25 MHz to 475.25 MHz.  If you look at the lower image/alias of your signal, you can see that the components below 467.25 MHz have diminishing power the further they are from 467.25 MHz.  In contrast, those same frequency components of the central (intended) image increase in power the further they are from 475.25 MHz.  This clearly indicates that the frequencies of the lower image are attenuated more the further they are from the passband.

The default baseband filter selection is 75% of the sample rate which is sufficient for moderate attenuation of aliases.  By changing that setting to 100% of the sample rate, you greatly diminished the attenuation of aliases.  For better attenuation of aliases I suggest that you synthesize your signal with a sample rate of 16 Msps, set the HackRF's sample rate to 16 Msps, and set the HackRF's baseband filter to 8 MHz (50%).

Videogamer555, 2020-01-25:
> Firstly, HackRF One does not have a bandpass filter. It does have a selectable baseband filter which you are interpreting to be a bandpass filter.
>
> Secondly, your screenshots indicate that the [...]

I thought the signal images were symatrical though. If there's an image below the lowest frequencies of the intended signal, then there should also be an image above the highest frequencies of the intended signal, and they should be equally distant from the intended signal. If one image starts at 4MHz below the intended signal, then the other image should start at 4MHz above the intended signal. Yet there was no image above the intended signal. That seems to indicate that the HackRF One's filter is doing a better job at attenuating the higher image frequencies than lower image frequencies. What's going on with that?

mossmann, 2020-01-25:
Aliases are not symmetrical (unless they are aliases of the negative image of real valued signals, which we do not have here).  You are correct that there is an alias above your intended signal.  That image should start at 478.5 MHz and should have its strongest peak at 479.25 MHz.  Because it starts 3.25 MHz away from the edge of the passband, however, it will be attenuated more than the portion of the lower alias you have observed (which starts right at the lower edge of the passband).

I suggest that you spend some time learning about DSP.  A good place to start is my video series, [Software Defined Radio with HackRF](https://greatscottgadgets.com/sdr/).

Videogamer555, 2020-01-26:
Interesting. The upper alias appears to be attenuated completely, because I couldn't see it at all in SDR#.
However, I have since tried your previous suggestion to use a sample rate 16MHz, and a baseband filter width of 8MHz , and that seems to work for completely removing the lower alias. I guess the thing is the TX baseband filter in the HackRF isn't a sharp "brick wall" type of filter.

closed 2020-02-03 by miek