greatscottgadgets/hackrf #701: Syncing phase with multiple SDR's for use in phased array

emfennelly, 2020-01-30:
Hi, 

I am having an issue trying to use two HackRF's to create a signal that is out of phase from the other for use in a phased array.  I am able to generate the desired phase shift through simulation in GNU radio but when measuring phase using a scope, it is a different story.

### Steps Taken
1. I used the built in CLKIN and CLKOUT ports and this did not work.
2. I used the instructions outlined in this wiki.
https://github.com/mossmann/hackrf/wiki/Multiple-device-hardware-level-synchronization

This synchronized the boards but from what I am reading and seeing, there is no GNU Radio support for this feature.  I am in the process of using hackrf_transfer in the shell to transmit the signals with the hope that this will remedy the phase issue.

Does anyone have any experience with this?  Any suggestions are much appreciated.

Thanks

jnunezortuno, 2020-01-30:
Hi emfennelly,

Did you read this conversation? 

https://github.com/mossmann/hackrf/pull/381

Perhaps it can help you.

Regards,
José

mossmann, 2020-02-06:
After tuning to a frequency and configuring the sample clock, there will be a random phase offset from one HackRF to another.  That phase offset should remain consistent if the HackRFs are clock synchronized and their tuning frequency and sample rate do not change, but the initial offset is unpredictable.  You'll need to calibrate.

closed 2021-01-25 by mossmann