greatscottgadgets/hackrf #703: invalid data at start of RX or TX

mossmann, 2020-02-05:
In RX mode (e.g. `hackrf_transfer -r`) the first USB buffer of data received by the host is often invalid, typically left over from a previous RX or TX operation.

In TX mode (e.g. `hackrf_transfer -t`) the first USB buffer of data transmitted over the air has the same problem.

mossmann, 2020-02-05:
For the specific case of the first RX operation after a HackRF is powered on, there is a trivial fix: set `phase = 1` as the initial value in hackrf_usb.c.

I want a more general solution than that.

mossmann, 2020-02-05:
sweep_mode() does the right thing in setting `phase = 1` when starting, but it ought to do something about  usb_bulk_buffer_offset as well, perhaps in baseband_streaming_enable().  All three modes (RX, TX, and sweep) should probably be mutually exclusive states in a state machine.

mossmann, 2020-02-06:
I have a branch in progress here: https://github.com/mossmann/hackrf/tree/transceiver

ebarnard, 2020-02-06:
See #662. Fixing this issue is a side effect of that PR.

ebarnard, 2020-02-06:
Your branch looks neater but it would be great if you could incorporate the changes to flush the SGPIO buffer and pre-fill the libusb tranfer buffers on transmit.

mossmann, 2020-02-06:
Ooh, I should have taken a closer look at #662 when I started investigating this issue.  I'm going through it now.

mossmann, 2020-02-11:
@miek Can you take a look at this too?

miek, 2020-02-11:
Looks good! Definitely tidies things up a lot.

I think it breaks the CPLD update though: CPLD update has been changed to a proper transceiver mode, but `usb_vendor_request_set_transceiver_mode` hasn't been changed to actually set the transceiver mode when `TRANSCEIVER_MODE_CPLD_UPDATE` is specified.

miek, 2020-02-12:
I rebased `transceiver` on master (here: https://github.com/miek/hackrf/tree/transceiver) and set up for some testing:

I'm tuned to 20MHz, sample rate 20M. I first transmit a 5MHz sine to fill the buffers, stop, then transmit a 0Hz sine and take a capture. CH1 is the TX LED (finally a reason to populate P5, I'm gradually filling all the empty headers :smiley:), CH2 is the RF output, CH3 is TXBBI+.

master (with a chunk of samples from the previous TX at the start):
[image]

miek/transceiver:
[image]

So it's certainly getting rid of that chunk of the previous samples, but the chunk of noise right after remains. I'll see if I can figure out where that's coming from.

ebarnard, 2020-02-12:
The chunk of noise is due to the fact that libhackrf sends four buffers containing uninitialised memory to the HackRF before the first user supplied buffer. It calls `libusb_submit_transfer` for each of the four libusb transfers without first filling the transfer buffers.

In #662 I've made the following changes to fix this:
1. libhackrf requests data to fill the four transfer buffers before initialising the transfers. Thus the first four buffers sent to the HackRF no longer contain the contents of uninitialised memory.
2. In TX mode the HackRF waits for the first USB buffer to be received before starting to output samples. This means that in hardware triggered mode the first sample to be output is the first user provided sample.

miek, 2020-02-12:
@ebarnard Yep, that was it - thank you! I've pushed a fix to my branch (I'd like to get `transceiver` merged to fix this issue first, then I'll look at getting your changes merged on top).

miek, 2020-02-12:
RX is also looking good, here's a capture of a 50.5MHz signal, take with `hackrf_transfer -r temp.cs8 -s 20e6 -f 50e6 -n 20000000 -l 16 -g 24`:

`master`:
[image]

`transceiver`:
[image]

Obviously there's still some junk, probably from PLLs settling. Not sure if we want to include that in the scope of this issue?

ebarnard, 2020-02-12:
I would imagine that could be down to hackrf_transfer calling `hackrf_set_freq` (and a few other setup functions) after `hackrf_start_tx/rx` [(link)](https://github.com/mossmann/hackrf/blob/master/host/hackrf-tools/src/hackrf_transfer.c#L966). 

`hackrf_set_freq` results in [`set_freq`](https://github.com/mossmann/hackrf/blob/master/firmware/common/tuning.c#L57) being called which first toggles the MAX2837 in and out of standby mode even if the frequency is not being changed via `max2837_set_mode(&max2837, MAX2837_MODE_STANDBY)` and `max2837_set_mode(&max2837, prior_max2837_mode)`.

I imagine this would have a unpredictable effects on the samples being read by the MAX5864 during this process.

It should be a trivial thing to fix assuming there's not a good reason for this ordering of setup operations.

PR #707 (2020-07-26): Clean up transceiver modes & fix invalid data on start of TX/RX
Fixes #703 

This rebases @mossmann's transceiver branch on master, and adds a few small fixes discussed in #703
closed 2020-07-26 by mossmann