greatscottgadgets/hackrf #705: HackRF one Power issue?

stoneburner, 2020-02-16:
### Steps to reproduce
1. i power on my hackrf - nothing happens

### Expected behaviour
it should get enumerated as usb device (again)

### Actual behaviour
he 3v3 led is on, the TX led is about 50% on, no other led is on
the device gets not recognized in normal mode or dfu mode
it worked fine before 

i have access to multimeters and oscilloscope

similar issue: #641 

(i am pretty sure i killed my hackrf somehow, although i only ever connected the ANT500 to it, so an answer like, "you broke it, sorry" is also fine i will order another one anyway)

timbakirov, 2020-03-02:
Same problem! Looks like the TPS62410 too weak (by current) for powering the entire circuit. Also, i think there is a general mistake in the schematic. TPS62410 DC-DC converter has two outputs- Vout1 and Vout2. It seems like  voltage on Vout1 should be no more than 1.5V. There is no explicit warnings about this in the datasheet, but look at the Table 3. Yes, this is a setting using registers, but there is no voltage in the table greater than 1.5 volts for converter 1. Also recommended range for R46 and R55 (used in voltage setting divider) is 180 kΩ to 360 kΩ. But in circuit  - 162 kΩ. Fortunately there are trace jumpers pads for connecting external supply. I use two TPS63020 (Vout 1.2 - 5.5V 4A) on tiny PCB.

stoneburner, 2020-07-05:
> Fortunately there are trace jumpers pads for connecting external supply. I use two TPS63020 (Vout 1.2 - 5.5V 4A) on tiny PCB.

can you give more details about where the jumper pads are? did you remove the old converter?

mossmann, 2021-01-26:
> Same problem! Looks like the TPS62410 too weak (by current) for powering the entire circuit. Also, i think there is a general mistake in the schematic. TPS62410 DC-DC converter has two outputs- [...]

We are using the TPS62410 within specifications.

mossmann, 2021-01-26:
@stoneburner Do you still have an issue with your HackRF One?

stoneburner, 2021-01-26:
no, i just replaced the TPS62410 with a new one, works fine again now, thanks!

closed 2021-01-26 by stoneburner