greatscottgadgets/hackrf #710: Due to a bad cable, the U15 LXES1TBCC2-004 burned out. Help me.

skyden-ua, 2020-03-04:
Due to a bad cable, the U15 LXES1TBCC2-004 burned out. Help me.
For earlier I apologize for English google translator is to blame for everything))

The story is this, I connected the device with a bad cable and after a few seconds smoke came out of it.
I know for sure that the U15 LXES1TBCC2-004 burned out because it had a hole.
the question is, what else could suffer?
a local mobile phone repairman, got drunk and said should earn without this detail (of course this did not happen)

I have a portapack without battery.
Could I supply power to the contacts to test the functionality of these devices without USB?
if so, what contacts and how many volts?

With great pleasure I will accept any advice on the diagnosis and restoration of these devices.

PS
Be careful with the cables, have a nice day everyone.

[image]
[image]

RChadwick7, 2020-03-23:
I'm in the same boat. The LXES1TBCC2-004 was discontinued, as I'm sure you know by now. The only substitute I could find was a DRTR5V0U4LP16. You should check the datasheets yourself, and please let us know if it works.

catalinalb, 2020-03-23:
As I read through all of this I start to wonder since I use a magnetic micro usb cable for the hackrf to avoid damaging the socket. It works fine still even with the portapack even though the usb cable is thin. 🙂

mossmann, 2020-03-24:
We use Vishay part number VBUS54CV-HSF-G4-08 for U15. This has been documented in the KiCad source since 2018.

mossmann, 2020-03-24:
Your HackRF One should operate normally (but lacks ESD protection) with U15 removed. Testing in that configuration would be a good first step to determine if everything else is working correctly.

miek, 2020-03-24:
Could you give a bit more info about how you were powering the HackRF when it went wrong? Make/model of any battery packs or power supplies would be helpful.

skyden-ua, 2020-03-31:
> Ваш HackRF One должен работать нормально (но без защиты от электростатического разряда) с удаленным U15. Тестирование в этой конфигурации было бы хорошим первым шагом, чтобы определить, все ли работает правильно.

Hello.
 Today I drank U15 LXES1TBCC2-004 and applied power to usb. Portapack showed no sign of life. The detail that I noted "t" is heated. I did not wait for a critical temperature and turned it off.
 I am also confused by the detail that I noted "?" it’s probably a stabilizer of drunkenness and a burn mark on it. Perhaps this is from U15 LXES1TBCC2-004 and the part is "?" whole.
Can I apply stabilized voltage to some place from the portapack side and run both boards? I met portapack with a built-in battery (I don’t have one) and pre-sented that you can revive them in this way.
If so, then what voltage do I need to apply and to which contacts?
I will be glad to any advice on the recovery and diagnosis of my accident.
[image]

skyden-ua, 2020-03-31:
> Не могли бы вы дать немного больше информации о том, как вы использовали HackRF, когда он пошел не так? Марка / модель любых аккумуляторов или блоков питания будет полезна.

It looked like there was a hidden mechanical damage inside the cable, and at some point a short circuit occurred. The cable was high-quality, but it worked for many years.

vasja2000, 2020-04-05:
> Hello.
>  Today I drank U15 LXES1TBCC2-004 and applied power to usb. Portapack showed no sign of life. The detail that I noted "t" is heated. I did not wait for a critical temperature and turned it [...]

I think it can help
[image]

vasja2000, 2020-04-05:
Кстати да! где этот транзистор или что это?
By the way, yes! where is this transistor or what is it?
[image]

mossmann, 2021-02-04:
> Can I apply stabilized voltage to some place from the portapack side and run both boards?

You can apply 5 V to P20 pin 22 and ground to P20 pin 19.

mossmann, 2021-02-04:
> Кстати да! где этот транзистор или что это?
> By the way, yes! where is this transistor or what is it?

The parts you have circled are resistors. One of them is a 0 ohm resistor, so it is really a short.

closed 2021-04-23 by straithe