greatscottgadgets/hackrf #712: hackrf broken after firmware update

dend0r, 2020-03-10:
### Steps to reproduce
So first i updated the firmware to 2018.01.1. No error messages on both steps spi and cpld. `hackrf_info` showed me the new firmware version but with `hackrf_transfer -r /dev/null` i got the error message "couldn't transfer any bytes for one second". Then i tried to flash an older firmware wich bricked it even more because now when i restart the hackrf `hackrf_info` says "No HackRF boards found." and just the 3v3 and the rf led are on. i also tried it again with the new firmware still the same error. as i write this i was checking the hackrf to make sure i get the error messages right and `hackrf_transfer -r /dev/null` seems to work after i dfu boot and use `dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu` but if i reset or unplug it and plug it in again just the 3v3 and the rf led are on

i think its like issue #380 but how do i know if its a hardware fault?

### Expected behaviour
It should restart properly

### Actual behaviour
after reset just the 3v3 and the rf led are on

### Version information
**Operating system**: 
 Linux Mint 19.3

**hackrf_info output:** 
hackrf_info version: 2018.01.1
libhackrf version: 2018.01.1 (0.5)
No HackRF boards found.

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

JonaldJohnston, 2020-03-15:
flashing via dfu only loads the firmware into ram, or so i believe. so when you restart, it goes away. to do a full firmware flash, flash via dfu and then use spi

dend0r, 2020-03-15:
sorry i wasn't clear enough. of course did i flash it like described [here](https://github.com/mossmann/hackrf/wiki/Updating-Firmware) first dfu, then spiflash

dend0r, 2020-05-01:
it was a hardware fault. nooelec sent me a new one

closed 2020-05-01 by dend0r

RoninActual, 2021-08-09:
It looks like I may have the same issue.  Hoping there is a fix now instead of having to replace the hardware?

`$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download        [=========================] 100%        35508 bytes
Download done.
dfu-util: unable to read DFU status after completion

$ hackrf_spiflash -w hackrf_one_usb.bin
File size 35444 bytes.
hackrf_open() failed: HackRF not found (-5)`

straithe, 2021-08-09:
Please open a new GitHub issue for your concerns.