greatscottgadgets/hackrf #714: TX File format

eoinoc87, 2020-03-13:
Can somebody please describe the file format needed for TX?

I would like to create my own samples in excel or from Matlab and I need to know what format is accepted.

This is what I have tried, but I don't think its working

I,Q
I,Q
I,Q

I and Q were in 8 bit ASCII format, aka readable text, must the samples be binary? Should I have a newline after every IQ sample pair?

Please let me know how I should format the samples :)

miek, 2020-03-13:
Yes, the samples must be binary (8-bit signed integers), with nothing else between samples.

eoinoc87, 2020-03-16:
Thanks for the reply, I have tried to format my file correctly, but I'm still not getting a nice RF waveform on the spectrum analyser. I have formatted the file now like this:
Byte0Byte1Byte2Byte3
     i        q      i        q

[binsamples.txt](https://github.com/mossmann/hackrf/files/4337556/binsamples.txt)

This file contains 4 periods of a sine wave in iq format with the wave having 8 samples per period.

I tried to transmit it with this command:

hackrf_transfer -t binsamples.txt -R -f 1243000000 -a 1 -x 25 -s 8000000

What am I doing wrong?

miek, 2020-03-24:
It looks like there's a bug in hackrf_transfer that means the repeat option doesn't work right with files smaller than 256kB - if you repeat your samples to make a file bigger than that, it should work right.

closed 2020-03-24 by miek