greatscottgadgets/hackrf #715: LXES1TBCC2-004 chip blown

JonaldJohnston, 2020-03-15:
my usb connection had been completely not working for a while (#681), couldn't figure out why. issue #713 led me to look at the LXES1TBCC2-004 chip on my board, and it looks like the smoke got let out: [image]

Anywhere i could get a replacement chip? Or should i just send it in? its listed as obsolete and im having trouble finding them for sale.

miek, 2020-03-24:
Could you give a bit more info about how you were powering the HackRF when it went wrong? Make/model of any battery packs or power supplies would be helpful.

JonaldJohnston, 2020-03-25:
I don't exactly know, i put the HackRF away for a while and brought it back out so i could update the portapack firmware, that's when i discovered that the USB was no longer working. i have no idea what cables/battery packs i was using.

JonaldJohnston, 2020-03-25:
Looks like #710 mentions using VBUS54CV-HSF-G4-08 instead of the LXES1TBCC2-004, will try.

JonaldJohnston, 2020-04-11:
Looks like taking out the LXES1TBCC2-004 without replacing it with anything is a temporary fix, although i wouldn't recommend it as you might end up damaging something you cant replace :), will mark as closed for now.

closed 2020-04-11 by JonaldJohnston