greatscottgadgets/hackrf #718: Unable to get HackRF One working

Allasso, 2020-03-23:
### Steps to reproduce
Install gnuradio, gr-osmosdr, hackrf, libhackrf-dev on Macos and Ubuntu
Install gqrx on Macos
Install antenna and connect USB on HackRF One
For gnuradio-companion, create various flow graphs per Michael Ossman tutorial, gnuradio wiki, run flow graphs.
For gqrx, open app, set for WBFM

See also forum post regarding this issue which include screen shots:
[https://forums.radioreference.com/threads/unable-to-get-hackrf-one-working-properly-on-macos.403521/](https://forums.radioreference.com/threads/unable-to-get-hackrf-one-working-properly-on-macos.403521/)

### Expected behaviour
[1]Rx light comes on.
[2]FFT plot appears.
[3]Set center frequency to known local FM station frequency and hear audio.
[4]On 2.4 GHz band, see change in FFT levels where moving 2.4 GHz access point closer/farther from HackRF One

### Actual behaviour
gqrx:
[1] above happens
[2] above happens
[3] above does not happen (no audio but white noise, except at center frequency, where white noise is squelched, but no audio.)
[4] does not happen (making signal stronger has no effect on FFT levels.)

gnuradio:
[1] above happens
[2] above happens on Macos, but no FFT plot on Ubuntu.
[3] above does not happen (no audio but white noise.)
[4] above does not happen (making signal stronger has no effect on FFT levels.)

### Version information
**Operating system**:
Macos 10.13.6, Ubuntu 18.04

**hackrf_info output:**

hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000057b068dc22262b63
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00664351

gnuradio version:
3.7.13.5 (Macos) 3.7.11 (Ubuntu)

gqrx version:
2.11.5 (Macos)

### Output

gnuradio-companion - Macos:

```
gr::fft: can't import wisdom from /Users/kevin/.gr_fftw_wisdom
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.13.5
built-in source types: file fcd rtl rtl_tcp uhd sdrplay hackrf bladerf rfspace airspy soapy
[INFO] [UHD] Mac OS; Clang version 10.0.0 (clang-1000.10.44.4); Boost_107100; UHD_3.15.0.0-MacPorts-Release
get_devices started
Device count: 0
get_devices end
Using HackRF One with firmware 2018.01.1
gr::log :INFO: audio source - Audio sink arch: osx
gr::log :INFO: audio_osx_sink0 -

Using output audio device 'Built-in Output'.
  ... which is the current default output audio device.
  Changing the default output audio device in the System Preferences will
  result in changing it here, too (with an internal reconfiguration).```
Insert any commandline or build output here
```
gnuradio-companion - Ubuntu:

```
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using HackRF One with firmware 2018.01.1
```

miek, 2020-03-24:
Which antenna are you using? Do you have another antenna you could try, too?

Allasso, 2020-03-24:
Interesting, I was just typing this out when your post came through.

After trying a dipole antenna rather than the ANT500, carefully adjusting settings, and using gqrx, I became aware that I am able to barely hear the most well-received FM station in my area.  Note that I live in a remote area and it is a distant station, however I was judging the results against what I am able hear on my vehicle's FM radio, where it comes in very clear and powerful.

Nonetheless, with the HackRF I am barely able to listen to the content among the white noise after cranking the audio gain up to 19.8 dB in gqrx.  I compared this to my RTL-SDR dongle, using the same antenna, and while I still have to crank it up, the content is much louder and more listenable.  (I also observe that with the HackRF I get -55 dBFS, whereas with the RTL-SDR I get -11 dBFS, although I don't know exactly how full-scale values are arrived at so don't know if that is a proper comparison.)  Note I was using WFM(mono) setting as it worked the best.

So it would seem that the HackRF is working (?) however if it is, I would like to understand why a particular FM station is barely listenable on HackRF, when it comes in solid and clear on my FM radio.  And also, while it isn't great on my RTL-SDR, it is so much better than the HackRF.

Allasso, 2020-03-24:
Note, the dipole I tried is tuned to the frequency of the referenced station (88.5 MHz, 63.5")

Allasso, 2020-03-25:
> I would like to understand why a particular FM station is barely listenable on HackRF, when it comes in solid and clear on my FM radio.

So is that expected behavior?  Do commercially made radios just do a much better job of discriminating weaker signals?

straithe, 2021-06-03:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-06-03 by straithe

roadmutt, 2021-06-04:
yes im still having the same issue is there somewhere i can just download a
firmware flash and or the SD card image and then just transfer the files
ive personally created to it afterwards