greatscottgadgets/hackrf #719: HackRF not powering on (probably power IC broken)

An0nymo0se, 2020-03-24:
### Steps to reproduce
1.Power Nooelec LNA
2.Plug in hackrf to computer
3.Bad luck.

### Expected behaviour
Hackrf Lights up, detected by computer.
### Actual behaviour
Area near USB port gets very hot. P1 and P8 short-circuited. No visible damage. No lights. 
Area that heats up shown below:

[image]

I'm pretty sure it's a lost cause, but I might as well try.

miek, 2020-03-24:
I assume you've tried it with the LNA disconnected - do you see the same result?
Also, just to make sure, are you saying that P1 is shorted to P8?

It might not be the power IC itself - that may just be getting very warm because of a short elsewhere on the board.

An0nymo0se, 2020-03-24:
P1 positive and GND is shorted. P8 positive and GND is also shorted. Also,
yes, I've tried with everything disconnected except the USB and the same
thing happened.

miek, 2020-03-24:
Are you measuring between the two pins on P1, or P1 to a GND pin elsewhere? The two pins on P1 are supposed to be shorted together. (and the same for P8)

An0nymo0se, 2020-03-24:
Oh, that's weird. Isn't P8 supposed to supply VCC? when I measure the
voltage between P8 and ground (another pad on the board) the voltage reads
about 0.7 volts.

miek, 2020-03-24:
Yeah, you should see VCC (3.3V) on P8 when everything's working right. Seeing 0.7V there (and the hot IC) suggests there is still a short elsewhere on the board.

The point of those headers is to be able to cut the short trace between the pins to isolate the power supply from the rest of the board. This can be useful for current measurement, or debugging.

You might want to investigate warranty options first with the vendor you purchased from. If you want to have a go at debugging this issue yourself, you could carefully cut the trace between the two pins on P8 and see if you see 3.3V on one side of it afterwards. That would tell you whether the fault is in the power supply, or a short elsewhere.

An0nymo0se, 2020-03-24:
Thanks for the advice. Unfortunately, I don't have a warranty. Also, would
it be possible to bypass the power supply and power the board externally
through j20 while only connecting the data pins for USB? Thanks for the
help.

bobofruit1970, 2020-07-07:
An0nymo0se, did You find what was shorted?
My board has the same problem.
Diodes no light up and hot power section around power chip U21.
Simple USB power meter shows curent 1,7 A.

My board died just after starting transmitting via walkie-talkie in the same room.
Is it possinle that this killed RX amplifier U13 and is now shorted?
Before I start to cut wires I would like to ask . Maybe someone  knows those symptoms and knows good this board.
What can be shorted in this board on 3.3V line?

miek, 2020-07-23:
Powering the board externally gets complicated, since it supplies both 3.3V and a switchable 1.8V. You're still left with the problem that there's probably a short on 3.3V.

I would still recommend cutting the small trace between the two P8 pins, then you can see whether the power supply works OK afterwards, and also power the 3.3V rail from an external current limited power supply to try and find the short. You can always install a header & jumper on P8 afterwards to reconnect it

closed 2020-08-18 by miek

andyparkers, 2021-11-12:
> Thanks for the advice. Unfortunately, I don't have a warranty. Also, would it be possible to bypass the power supply and power the board externally through j20 while only connecting the data pins for USB? Thanks for the help.
> […](#)

It seems I have the same problem. Have you solved yours?

An0nymo0se, 2021-11-12:
I haven't solved mine yet. I've looked everywhere for something that might be shorting VCC but couldn't find anything. I'm thinking of desoldering the pin headers because I _think_ i might've messed with them a while back. Other than the pin headers, I can't really see anything that could be causing issues. I should also note that the board I bought was a cheap knockoff (i think, based on what the board looks like and price I paid). I'll post an update when I get around to doing that.