greatscottgadgets/hackrf #721: Multiple frequency-transmission on HackRF

elainegrace, 2020-03-25:
Hi! Does anyone here know how to simultaneously/consecutively transmit into multiple (more than one) radio frequencies using HackRF One? In my process I am only able to transmit to a single frequency by running this flowgraph into GNU Radio (attached is a screenshot) 
[image]

WoJiaoLeimen, 2020-06-01:
I also encountered the same problem. It is possible to simultaneously transmit multiple radio frequency using HackRF One? If not, it is possible to create a separate GNU radio flowgraph files and then transmit it one by one?

mossmann, 2021-11-12:
HackRF One can transmit many signals at once provided that the total bandwidth of all those signals together is less than 20 MHz. In the example flowgraph above, you could transmit several wideband FM channels by adding multiple source blocks, each with its own WBFM Transmit and resampler blocks. Then you would need to shift the frequency of each transmission (I suggest doing this with a Frequency Xlating FFT Filter) and add all the transmissions together with an Add block.

The same process in reverse (listening to several FM radio stations simultaneously) is discussed in my video series: https://greatscottgadgets.com/sdr/

closed 2021-11-12 by mossmann