greatscottgadgets/hackrf #722: HackRF in the role of an oscilloscope

Rashid-S, 2020-03-26:
Someone tried to use a HackRF in the role of an oscilloscope in a GNU Radio. Share your experience please.

GravesJake, 2020-04-02:
It is possible to use a HackRF as an "oscilloscope" but it will be very easy to damage the ADC. Check the SDR video series for an example.

closed 2020-04-06 by GravesJake

ktemkin, 2020-04-07:
I'd definitely note the quotes on there-- there are a bunch of limitations that make HackRF hard to use this way (including AC input coupling and a relatively-low-impedance frontend that's meant for small amount of RF power). :)

5ch4um1, 2021-01-14:
Sort of, i tried to look at OOK radio signals with the time sink in gnuradio 3.7:
[image]
The closest i ever got to using an oscilloscope was running xoscope (soundcard o-scope) on my laptop, so i am a total beginner with this topic, but if you activate the controls and the grid, the time sink looks a bit like an oscilloscope.
But while the trigger function in xoscope worked for me more or less intuitively as expected, i could never figure out how this is supposed to work in gnuradio, probably i am using it wrong? It just won't stop?
[image]