greatscottgadgets/hackrf #727: HackRF Not Responding aftre transmit.

letvinov28, 2020-04-01:
HostOS:Win10
VirtualOS:ParrotOS
My hackRF stop work after first trasmit.

> └──╼ $hackrf_transfer -r /dev/null -s 20000
call hackrf_set_sample_rate(20000 Hz/0.020 MHz)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
 0.5 MiB / 1.000 sec =  0.5 MiB/second
 0.3 MiB / 1.004 sec =  0.3 MiB/second
^CCaught signal 2
 0.3 MiB / 0.510 sec =  0.5 MiB/second

Exiting...
Total time: 2.51408 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
┌─[user@parrot]─[~]
└──╼ $hackrf_transfer -r /dev/null -s 20000
call hackrf_set_sample_rate(20000 Hz/0.020 MHz)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
 0.0 MiB / 1.001 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 1.00116 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
┌─[✗]─[user@parrot]─[~]
└──╼ $

And in GQRX my hackrf stop work if i change any settings.

GravesJake, 2020-04-27:
Can you describe the GQRX issue? Paste `hackrf_info` output?

straithe, 2021-01-26:
Are you still having trouble? We will close this issue if we don't get a reply within the next month.

letvinov28, 2021-01-26:
Hello, last time when i test HackRF yes i had troubles. Today i will test
one more time and then i will reply you.

straithe, 2021-02-11:
Were you able to test your HackRF again?

letvinov28, 2021-02-11:
Please excuse me, I completely forgot that I had to do this. There were
many unexpected problems. I promise today by 3 o'clock (+ 0GMT) I will do
all the tests. But as far as I remember, the problem was only in Vmware on
the physical host, everything worked well.

letvinov28, 2021-02-11:
Hello, I checked my device and as I said the problem is only on this
machine.

straithe, 2021-02-16:
What happens when you try a different OS, such as Ubuntu, on your VM?

letvinov28, 2021-02-16:
Same story , tested on Ubuntu, Arch , Debian. Maybe problem is in host os?
Mac OS

mossmann, 2021-03-02:
Thank you for bringing this to our attention, but unfortunately we are not able to support operation on virtual machines. There are too many combinations of vm software and operating systems for us to test, and USB pass-through is one of the least reliable features of vm software in our experience.  If you experience similar problems when operating in a native OS, please open a new issue.

closed 2021-03-02 by mossmann