greatscottgadgets/hackrf #730: Frequency generator

markl17, 2020-04-12:
### Steps to reproduce
1.i would like to make a frequency generator out of jacked
2.should have resolution of1.123456789 hz
3.should have a time setting for how fast to count should be able to generate to and from until 6ghz

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

miek, 2020-04-23:
Is there a specific issue you're running into? What information are you looking for?

markl17, 2020-04-23:
I want to get touch to work on my yoga 2 pro hd4400 4510

straithe, 2021-01-26:
Are you still experiencing this issue?

markl17, 2021-01-26:
yes

markl17, 2021-01-29:
A question I have a frequency counter used to be off a little but
exceptable when but signal emplifier signal reading remained the same but
now the counter telling me I am outputting 14ghz when all I am putting is
750mhz what happened did something getting burned because without amplified
it is pretty close to 750mhz

markl17, 2021-01-31:
hi can you guys explain to me why this thing used to work the amplifiers
used to not interfere with my counter and now I dont know if counter is
right or hackrf is somehow delugional this is a brand new one

mossmann, 2021-03-02:
The only signal generator software I know of that might meet your needs is `osmocom_siggen` from the [gr-osmosdr](https://osmocom.org/projects/gr-osmosdr/wiki) project. If that does not meet your needs, you would have to write your own signal generator software.

It is highly unlikely that even a faulty HackRF One is capable of producing significant energy at 14 GHz. I have done testing up to 7 GHz, and emission strength drops dramatically above 6 GHz. I am closing this issue as I believe I have answered your original question. If you would like further help about another matter, please open another issue with a clear description of your problem and what we can do to try to reproduce the problem.

closed 2021-03-02 by mossmann

markl17, 2021-03-02:
Thank you so much for your attention