greatscottgadgets/hackrf #732: unable to make hackrf_usb firmware

oldark87, 2020-04-15:
### Steps to reproduce
1. download gcc-arm-none-eabi, follow readme instruction to extract, install, and add to PATH
2. checkout libopencm3 https://github.com/mossmann/libopencm3 and run 'make'
3. cd to hackrf/firmware/hackrf_usb
4. mkdir build
5. cd build
6. cmake ..
7. make

### Expected behaviour
hackrf_usb.bin generated to then be loaded through spiflash

### Actual behaviour
make command errors out

### Version information
**Operating system**: ubuntu 18.04.4 LTS

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000057b068dc2578bd63
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x006c474f

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```

Scanning dependencies of target hackrf_usb_m0.elf
[  2%] Building C object CMakeFiles/hackrf_usb_m0.elf.dir/home/timms/hackrf/firmware/common/m0_sleep.c.obj
as: unrecognized option '-mcpu=cortex-m0'
CMakeFiles/hackrf_usb_m0.elf.dir/build.make:62: recipe for target 'CMakeFiles/hackrf_usb_m0.elf.dir/home/timms/hackrf/firmware/common/m0_sleep.c.obj' failed
make[2]: *** [CMakeFiles/hackrf_usb_m0.elf.dir/home/timms/hackrf/firmware/common/m0_sleep.c.obj] Error 1
CMakeFiles/Makefile2:168: recipe for target 'CMakeFiles/hackrf_usb_m0.elf.dir/all' failed
make[1]: *** [CMakeFiles/hackrf_usb_m0.elf.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

miek, 2020-04-20:
libopencm3 should not be checked out separately, it's included as a submodule within the hackrf repository.

You may have been working from instructions in the wiki that were out of date, the instructions in `firmware/README` are correct: https://github.com/mossmann/hackrf/blob/master/firmware/README

oldark87, 2020-04-21:
Hey @miek.  I tried following the steps on that page. git submodule init and update don't provide any feedback but no errors either.  I then ran git submodule foreach and was able to verify that two submodules were there. 

However I still see the exact same error when I attempt to run the make command (I did destroy my build dir and recreate it just in case - no change).

miek, 2020-04-21:
Under `firmware/hackrf_usb/build` could you run `make clean`, then paste me the output of `make VERBOSE=1`?

oldark87, 2020-04-21:
In case I had mucked things up I deleted the repo and checked it out fresh and am now getting different results. 

- First I verify the arm-none-eabi is installed by trying to invoke it by 'arm-none-eabi-gcc'.  Correctly reports that there are no input files.

-  clone the repo into my install directory

- cd into /hackrf and both init and update the submodules > success output message

- mkdir build inside /firmware/hackrf_usb.  'cd build', 'cmake ..'

- results look like it's hunting for my toolchain in the wrong place?

`~/hackrf/firmware/hackrf_usb/build$ cmake ..
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /
-- The C compiler identification is unknown
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- broken
CMake Error at /usr/share/cmake-3.10/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "/usr/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/timms/hackrf/firmware/hackrf_usb/build/CMakeFiles/CMakeTmp
   
    Run Build Command:"/usr/bin/make" "cmTC_c8ea9/fast"
    /usr/bin/make -f CMakeFiles/cmTC_c8ea9.dir/build.make CMakeFiles/cmTC_c8ea9.dir/build
    make[1]: Entering directory '/home/timms/hackrf/firmware/hackrf_usb/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_c8ea9.dir/testCCompiler.c.obj
    /usr/bin/arm-none-eabi-gcc    -o CMakeFiles/cmTC_c8ea9.dir/testCCompiler.c.obj   -c /home/timms/hackrf/firmware/hackrf_usb/build/CMakeFiles/CMakeTmp/testCCompiler.c
    as: unrecognized option '-meabi=5'
    CMakeFiles/cmTC_c8ea9.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_c8ea9.dir/testCCompiler.c.obj' failed
    make[1]: *** [CMakeFiles/cmTC_c8ea9.dir/testCCompiler.c.obj] Error 1
    make[1]: Leaving directory '/home/timms/hackrf/firmware/hackrf_usb/build/CMakeFiles/CMakeTmp'
    Makefile:126: recipe for target 'cmTC_c8ea9/fast' failed
    make: *** [cmTC_c8ea9/fast] Error 2
   

 

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:24 (project)

-- Configuring incomplete, errors occurred!
See also "/home/timms/hackrf/firmware/hackrf_usb/build/CMakeFiles/CMakeOutput.log".
See also "/home/timms/hackrf/firmware/hackrf_usb/build/CMakeFiles/CMakeError.log".`

oldark87, 2020-04-23:
Trying this a few more time from the beginning. I'm not sure what I'm doing different but something got me back to the original error. Here's the verbose output that you requested.

```
make clean
:~/hackrf/firmware/hackrf_usb/build$ make VERBOSE=1
/usr/bin/cmake -H/home/timms/hackrf/firmware/hackrf_usb -B/home/timms/hackrf/firmware/hackrf_usb/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/timms/hackrf/firmware/hackrf_usb/build/CMakeFiles /home/timms/hackrf/firmware/hackrf_usb/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/timms/hackrf/firmware/hackrf_usb/build'
make -f CMakeFiles/hackrf_usb_m0.elf.dir/build.make CMakeFiles/hackrf_usb_m0.elf.dir/depend
make[2]: Entering directory '/home/timms/hackrf/firmware/hackrf_usb/build'
cd /home/timms/hackrf/firmware/hackrf_usb/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/timms/hackrf/firmware/hackrf_usb /home/timms/hackrf/firmware/hackrf_usb /home/timms/hackrf/firmware/hackrf_usb/build /home/timms/hackrf/firmware/hackrf_usb/build /home/timms/hackrf/firmware/hackrf_usb/build/CMakeFiles/hackrf_usb_m0.elf.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/timms/hackrf/firmware/hackrf_usb/build'
make -f CMakeFiles/hackrf_usb_m0.elf.dir/build.make CMakeFiles/hackrf_usb_m0.elf.dir/build
make[2]: Entering directory '/home/timms/hackrf/firmware/hackrf_usb/build'
[  2%] Building C object CMakeFiles/hackrf_usb_m0.elf.dir/home/timms/hackrf/firmware/common/m0_sleep.c.obj
/usr/bin/arm-none-eabi-gcc  -I/home/timms/hackrf/firmware/hackrf_usb/../../firmware/libopencm3/include -I/home/timms/hackrf/firmware/hackrf_usb/../../firmware/common  -std=gnu99 -Os -g3 -Wall -Wextra -DHACKRF_ONE -DLPC43XX -DLPC4320 -DTX_ENABLE -D'VERSION_STRING="git-687c72f"'  -fno-common -MD -mthumb -mcpu=cortex-m0 -mfloat-abi=soft -DLPC43XX_M0 -o CMakeFiles/hackrf_usb_m0.elf.dir/home/timms/hackrf/firmware/common/m0_sleep.c.obj   -c /home/timms/hackrf/firmware/common/m0_sleep.c
as: unrecognized option '-mcpu=cortex-m0'
CMakeFiles/hackrf_usb_m0.elf.dir/build.make:62: recipe for target 'CMakeFiles/hackrf_usb_m0.elf.dir/home/timms/hackrf/firmware/common/m0_sleep.c.obj' failed
make[2]: *** [CMakeFiles/hackrf_usb_m0.elf.dir/home/timms/hackrf/firmware/common/m0_sleep.c.obj] Error 1
make[2]: Leaving directory '/home/timms/hackrf/firmware/hackrf_usb/build'
CMakeFiles/Makefile2:168: recipe for target 'CMakeFiles/hackrf_usb_m0.elf.dir/all' failed
make[1]: *** [CMakeFiles/hackrf_usb_m0.elf.dir/all] Error 2
make[1]: Leaving directory '/home/timms/hackrf/firmware/hackrf_usb/build'
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

miek, 2020-04-23:
It does look like an issue with the toolchain installed. Could you paste the output of `/usr/bin/arm-none-eabi-gcc -v`?

oldark87, 2020-04-23:
```
/usr/bin/arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=/usr/bin/arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/6.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: ../src/configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:6.3.1+svn253039-1build1 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=rmprofile CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip
Thread model: single
gcc version 6.3.1 20170620 (15:6.3.1+svn253039-1build1) 
```

miek, 2020-04-23:
OK, it's picking up the Ubuntu packaged version of `arm-none-eabi-gcc` instead of the one you downloaded and installed. This is because the instructions given (`export PATH=$PATH:$install_dir/gcc-arm-none-eabi-*/bin`) are ordered to prioritise the system paths.

Swapping it around and doing `export PATH=$install_dir/gcc-arm-none-eabi-*/bin:$PATH` should fix it. Keep in mind you'll need to put that command in `~/.bashrc` or similar to apply it every time you open a terminal.

oldark87, 2020-04-29:
Resolved. Thank you for the assistance!

closed 2020-04-29 by oldark87