greatscottgadgets/hackrf #733: Kernel problem: device descriptor read/64, error -110

bialybudyn, 2020-04-17:
### Steps to reproduce
1.Configure autosuspend like in instruction
2.Downgrade USB3.0 to USB2.0 in BIOS
3.Plug HackRF One
4.Start hackrf_info for the first time(worked)
5.Start hackrf_info for the second time(not responding)
6.Try to restart board by button

### Expected behaviour
Standard working HackRF

### Actual behaviour
The board was stop working after few calls(hackrf_info or from other 3party programs). The USB LED stop after push reset(and only after that) and I can't access to board.  
My first guess was cable but i plug it in another computer and HackRF worked. After that i make whole thing with TLP and autosuspend and it works(hackrf_info shows device info but only for the first time). After that I found on radioamateur forums that changing USB 3.0 support to USB 2.0 support in BIOS helps.(that's also doesn't work for me). 

### Version information
**Operating system**:
5.4.0-kali4-amd64 #1 SMP Debian 5.4.19-1kali1 (2020-02-17) x86_64 GNU/Linux

**hackrf_info output:**

**First call:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc29569a5f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00664768
**Second call**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc29569a5f
Board ID Number: 2 (HackRF One)
**sometimes stops on Serial number and after force quit ctrl-c - Third time**

hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
hackrf_board_id_read() failed: Input/Output Error (-1000)
                                           
### Output
```

**tlp-stat**
+++ PCIe Active State Power Management
/sys/module/pcie_aspm/parameters/policy = [default] performance powersave powersupersave

+++ Runtime Power Management
Device blacklist = (not configured)
Driver blacklist = amdgpu mei_me nouveau nvidia pcieport radeon

/sys/bus/pci/devices/0000:00:00.0/power/control = on   (0x060000, Host bridge, ivb_uncore)
/sys/bus/pci/devices/0000:00:02.0/power/control = on   (0x030000, VGA compatible controller, i915)
/sys/bus/pci/devices/0000:00:16.0/power/control = on   (0x078000, Communication controller, mei_me)
/sys/bus/pci/devices/0000:00:1a.0/power/control = on   (0x0c0320, USB controller, ehci-pci)
/sys/bus/pci/devices/0000:00:1b.0/power/control = on   (0x040300, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:1c.0/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.1/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.2/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.3/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.6/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1d.0/power/control = on   (0x0c0320, USB controller, ehci-pci)
/sys/bus/pci/devices/0000:00:1f.0/power/control = on   (0x060100, ISA bridge, lpc_ich)
/sys/bus/pci/devices/0000:00:1f.2/power/control = on   (0x010601, SATA controller, ahci)
/sys/bus/pci/devices/0000:00:1f.3/power/control = on   (0x0c0500, SMBus, i801_smbus)
/sys/bus/pci/devices/0000:01:00.0/power/control = on   (0x020000, Ethernet controller, igb)
/sys/bus/pci/devices/0000:02:00.0/power/control = on   (0x020000, Ethernet controller, igb)
/sys/bus/pci/devices/0000:03:00.0/power/control = on   (0x020000, Ethernet controller, igb)
/sys/bus/pci/devices/0000:04:00.0/power/control = on   (0x020000, Ethernet controller, igb)
/sys/bus/pci/devices/0000:05:00.0/power/control = on   (0x028000, Network controller, bcma-pci-bridge)

+++ USB
Autosuspend         = disabled
Device whitelist    = (not configured)
Device blacklist    = 1d50:604b 1d50:6089 1d50:cc15 1fc9:000c
Bluetooth blacklist = disabled
Phone blacklist     = disabled
WWAN blacklist      = disabled

Bus 002 Device 012 ID 12d1:1573 control = on,   autosuspend_delay_ms = -1000 -- Huawei Technologies Co., Ltd.  (cdc_mbim, option)
Bus 002 Device 009 ID 0d8c:0014 control = on,   autosuspend_delay_ms = -1000 -- C-Media Electronics, Inc. Audio Adapter (Unitek Y-247A) (snd-usb-audio, usbhid)
Bus 002 Device 008 ID 0403:6001 control = on,   autosuspend_delay_ms = -1000 -- Future Technology Devices International, Ltd FT232 Serial (UART) IC (ftdi_sio)
Bus 002 Device 013 ID 413c:2113 control = on,   autosuspend_delay_ms = -1000 -- Dell Computer Corp.  (usbhid)
Bus 002 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms = -1000 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms = -1000 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms = -1000 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms = -1000 -- Linux Foundation 2.0 root hub (hub)

**lsusb**                                                                                                                                                         ✔ │ with root@commbox │ at 16:08:16   
Bus 002 Device 012: ID 12d1:1573 Huawei Technologies Co., Ltd. 
Bus 002 Device 009: ID 0d8c:0014 C-Media Electronics, Inc. Audio Adapter (Unitek Y-247A)
Bus 002 Device 008: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 002 Device 055: ID 1d50:6089 OpenMoko, Inc. Great Scott Gadgets HackRF One SDR
Bus 002 Device 013: ID 413c:2113 Dell Computer Corp. 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

[ 2892.447431] usb 2-1.2: new high-speed USB device number 52 using ehci-pci
[ 2892.567961] usb 2-1.2: device descriptor read/all, error -71
[ 2893.115377] usb 2-1.2: new high-speed USB device number 53 using ehci-pci
[ 2893.619336] usb 2-1.2: device not accepting address 53, error -71
[ 2893.619487] usb 2-1-port2: unable to enumerate USB device (**this happens after plug out/in to USB port**)
[ 2894.003301] usb 2-1.2: new high-speed USB device number 54 using ehci-pci
[ 2894.033257] usb 2-1.2: New USB device found, idVendor=1d50, idProduct=6089, bcdDevice= 1.02
[ 2894.033264] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[ 2894.033268] usb 2-1.2: Product: HackRF One
[ 2894.033271] usb 2-1.2: Manufacturer: Great Scott Gadgets
[ 2894.033274] usb 2-1.2: SerialNumber: 000000000000000087c867dc29569a5f
[ 2894.060099] usb 2-1.2: can't set config #1, error -71(**after few moments**)
[ 2894.294008] usb 2-1.2: USB disconnect, device number 54
[ 2894.515249] usb 2-1.2: new high-speed USB device number 55 using ehci-pci
[ 2894.544648] usb 2-1.2: New USB device found, idVendor=1d50, idProduct=6089, bcdDevice= 1.02
[ 2894.544656] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[ 2894.544660] usb 2-1.2: Product: HackRF One
[ 2894.544664] usb 2-1.2: Manufacturer: Great Scott Gadgets
[ 2894.544666] usb 2-1.2: SerialNumber: 000000000000000087c867dc29569a5f

```

GravesJake, 2020-04-27:
`hackrf_info` version is unknown in both called. Can you try updating your host tools to match your firmware version?

straithe, 2021-01-28:
Are you still experiencing this issue? We will close this issue if we don't get a reply within the next month.

bialybudyn, 2021-01-29:
Hi guys, 
the problem solved itself. Unfortunately I don't know why.  @GravesJake that was probably it but I made few reinstallations and sys updates after that and it was run. 

What was significant in my opinion - I purge hackrf tools then I installed libfttw3-dev and libusb from .deb packages and after that I finally installed hackrf tools from repo 

But I'm not sure that was the problem

closed 2021-01-29 by bialybudyn