greatscottgadgets/hackrf #734: i can include a screen shot , because I haven't  started yet to install hackrf one ! This page is the only on I found to install hackfr on ubuntu 18-04  ! That's why I want to understand this advice to follow it after following pelixm ...

zen1963, 2020-04-19:
i can't  include a screen shot , because I haven't  started yet to install hackrf one ! This page is the only on I found to install hackfr on ubuntu 18-04  ! That's why I want to understand this advice to follow it after following pelixm ... 
Thanks for you help ! I am a beginner !

_Originally posted by @zen1963 in https://github.com/mossmann/hackrf/issues/605#issuecomment-616197760_

ChasJutaro, 2020-04-19:
> i can't include a screen shot , because I haven't started yet to install hackrf one ! This page is the only on I found to install hackfr on ubuntu 18-04 ! That's why I want to understand this advice to follow it after following pelixm ...
> Thanks for you help ! I am a beginner !
> 
> _Originally posted by @zen1963 in [#605 (comment)](https://github.com/mossmann/hackrf/issues/605#issuecomment-616197760)_

Do you have the hackrf one device with you?
If you are new to Linux in general, I wouldn't start with the HackRF one, since using the HackRF One needs some knowledge on how to configure Linux.

If you're not a Linux beginner, then you can proceed and follow the guide in this repo's wiki on setting up on Ubuntu.

If you're a Linux beginner, follow the same guide but use a RTL-SDR dongle instead, cause that would be much easier to work with.

zen1963, 2020-04-20:
I Have the kackrf one device with me . 
And I tried ton install it following these orders :
Step 1 sudo apt-get update -y
Step 2
sudo apt-get install -y hackrf
At the end I find that ;
[image]
And when I triy to use this command : hackrf_info 
I find that on the terminal :
[image]
What does it mean ?

zen1963, 2020-04-20:
However the hackrf one is well connected : all the leds light up well !

zen1963, 2020-04-20:
What  can i do to unlock : ? I can access to the the hackrf info !

miek, 2020-04-20:
Check what groups your user is already in with the `groups` command. You probably need to add your user to the `plugdev` group if it's not listed, see here for details on how to do that: https://askubuntu.com/a/79566

zen1963, 2020-04-20:
How can I check " what  groups your user is already in with the groups command "
I can't access to hackrf info but I haven't any problem to access hackrf _transfer ! What does it mean ? 
Hackrf is well installed and I can use it even I can't access to hackrf _info ?

zen1963, 2020-04-20:
[image]

ChasJutaro, 2020-04-20:
ur hackrf_info is working correctly.
You just need admin access to access hackrf one device go retrieve the relevant information.
I.e. run again using "sudo hackef_info"

[Capture d’écran_2020-04-20_16-41-25]<https://user-images.githubusercontent.com/63970730/79767433-9da5ad80-8329-11ea-809c-7b9387f3479f.png>

miek, 2020-04-20:
> How can I check " what groups your user is already in with the groups command "

Type the word `groups` into a terminal. It will list which groups your user is in. If `plugdev` is not in the list, follow the instructions above to add it.

> I can't access to hackrf info but I haven't any problem to access hackrf _transfer ! What does it mean ?

`hackrf_transfer` is just showing the help there, it's not trying to access the USB device so it doesn't hit the permissions error

zen1963, 2020-04-20:
Thant you Very much for your help !  It was very helpful from you for helping me ! Have a nice day and more ....

zen1963, 2020-04-24:
Hi , 
I have  another problem when I enter this command ; hackrf_transfer   ͞  r  /dev/null it doesn't work as on the lesson 5 of Michael Ossman ! 
The  same lines with the command : hackrf_transfer are repeated !
And I notice that there is two lines that we don't find on the lesson 5 of Michael Ossman :
[image]
[image]

zen1963, 2020-04-26:
Any one to help me ?

zen1963, 2020-04-26:
Hi , What's wrong when I enter this line of command : hackrf_transfer  ͞  r /dev/null  it doesn't work ?

miek, 2020-04-27:
Could you paste the output from when you run `hackrf_transfer -r /dev/null`? The pictures above only show runs with no arguments.

zen1963, 2020-04-28:
Hi , What's wrong when I enter this line of command : hackrf_transfer  ͞  r /dev/null  it doesn't work ? 
[image]

zen1963, 2020-04-28:
I paste the output with arguments  ! It's what you asked ?

zen1963, 2020-04-28:
as I told it the runs it's the same when I enter this argument ; hackrf_transfer !

miek, 2020-04-28:
The symbol before 'r' must be '-' (a hyphen). The command you're entering has some other symbol which is not recognised.

zen1963, 2020-04-29:
Thanks Mike ! It's better but now when I run hackrf_transfer -r dev/null it fail to open file !
[image]

miek, 2020-04-29:
It should be `/dev/null` - note the leading slash.

zen1963, 2020-04-30:
Thank you so much Mike . I'm ashamed to be so dizzy !
When I run hackrfè_transfer -r /del/null it works now but  it doesn't stop !!! 
[image]
It contines without stopping !

miek, 2020-04-30:
Yep, that's expected - it will carry on until stopped. You can stop it with CTRL+C.

zen1963, 2020-05-02:
Thanks a lot Mike .. I am very grateful to you .
it works !! 
It means now everything is ok ?

closed 2020-09-09 by miek