greatscottgadgets/hackrf #737: No HackRF boards found.

ChansimDeng, 2020-05-13:
### Steps to reproduce
**problem：**
hackrf_info:
hackrf_info version: git-43e6f99
libhackrf version: git-43e6f99 (0.5)
No HackRF boards found.

1.my env:
Physical machine：mac os
virtual machine：ubuntu 16(hackrf use)

and I also successfully installed the required libraries in ubuntu, eg:libusb-1.0-0-dev hackrf

ChansimDeng, 2020-05-13:
@mossmann

ChansimDeng, 2020-05-13:
and command (lsusb) in ubuntu info:

Bus 001 Device 002: ID 0e0f:000b VMware, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 005: ID 0e0f:0008 VMware, Inc. 
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

GravesJake, 2020-05-13:
Are you able to see your device on the host machine?

ChansimDeng, 2020-05-14:
yeah, On my mac i can see usb information
[image]

GravesJake, 2020-06-01:
Sorry for the delay, I've not been able to reproduce this. What macos version are you running? VM software?

closed 2020-09-09 by miek