greatscottgadgets/hackrf #741: GNSS-SDR Not Acquiring

AgileEngineeringLLC, 2020-05-28:
Hi,
I'm trying to get GNSS-SDR running on the HackRF One. I think I'm getting closer, but still not getting acquisition. I installed a NooElec TXCO which seemed to help a little, but I'm still getting Loss of Channel Lock messages. I'm using a really high quality GPS antenna as well. I tried varying the gains, varying the pll and dll parameters, performing the calibration routine, and no luck.

Here is some sample output:

Current receiver time: 11 min 39 s
New GPS NAV message received in channel 1: subframe 5 from satellite GPS PRN 05 (Block IIR-M)
New GPS NAV message received in channel 3: subframe 5 from satellite GPS PRN 25 (Block IIF)
Current receiver time: 11 min 40 s
Loss of lock in channel 6!
Current receiver time: 11 min 41 s
Current receiver time: 11 min 42 s
Current receiver time: 11 min 43 s
Current receiver time: 11 min 44 s
Current receiver time: 11 min 45 s

I'm using the stock conf file for HackRF One from the GNSS-SDR Configuration page. Also, I'm able to get a good nav solution using their example data, so I know GNSS-SDR is running.

Appreciate any thoughts!

AgileEngineeringLLC, 2020-05-29:
Never mind.. got it working. Issue was that my gains were too high. Working great now.

miek, 2020-05-29:
Great! Glad you got it working.

closed 2020-05-29 by miek