greatscottgadgets/hackrf #742: How to implement Multiple Frequency-Transmission on HackRF

WoJiaoLeimen, 2020-06-01:
Good day! It is possible to simultaneously transmit multiple radio frequency using HackRF One? How? If not, it is possible to create a separate GNU radio flowgraph files and then transmit it one by one? I was able to transmit to a single frequency ( see the attached file). Did I miss something on my flowgraph in order to transmit in multiplt radio frequency? 
[image]

CvMaRTY, 2020-10-20:
This is from GNURadio:
[image]

I tried a different approach though. And this is because the WBFM transmit cuts everything over 16KHz, but I wanted to input MPX signal so it can also have Stereo & RDS. Both frequencies had very good modulation, but I was worried about the harmonics that I haven't solved yet.
[image]

straithe, 2021-02-02:
Do you still need help with your question?

closed 2021-03-03 by straithe