greatscottgadgets/hackrf #743: Operacake antenna switching

oldark87, 2020-06-01:
It's mentioned in one of your demo's that the hackrf supports a method to rapidly rotate the antennas through an operacake board.  Is there more documentation somewhere on how to accomplish this? 

Currently I'm trying to just run a script that just runs the command  'hackrf_operacake -o -a x' where x loops from 0-3 as fast as the processor will allow it to run.  There are two issues I see with this
1. Trying to run the df-mic-demo flowgraph with this script going results in errors that the hackrf is busy. Presumably it can't be sent these commands while also doing anything else.
2. No control over the rotation frequency.

In the Schmoocon 18 demo I can see that you're running a script file to trigger the rotation but how that is working is not shown or explained anywhere that I've been able to find.

GravesJake, 2020-06-01:
I'm not sure what script was used at Shmoocon, but I think at least some delay is needed to accomplish this. [This](https://github.com/docprofsky/hackrf/commit/ea31f86eafe863180f3785c5433d387c6c425f4b) might be helpful.

oldark87, 2020-06-04:
Thank you for the suggestion. I have compiled and flashed the firmware image from docprofsky's sctimer branch.  What I don't see is how to find what the new command line args that I need to use in order to trigger the rotator setup.  

I tried rebuilding the /host directory from his branch as well in case that would enable the command line help info but it fails with 
```undefined reference to symbol 'log2f@@GLIBC_2.26'
//lib/x86_74-linux-gnu/libm.so.6: error adding symbols: DSO missing from command line```

GravesJake, 2020-06-15:
Sorry for the late response, are you still having trouble finding the command? I'm personally not familiar with that code but can do some looking around if you haven't found it.

oldark87, 2020-06-16:
I'm not positive that I'm reading it correctly but it looks as though it's supposed to be non-configurable (outside of directly modifying the firmware and reinstalling it) and is activated anytime the operacake is set to have both pins on port 0 based off of the comment starting at line 68 of the file you linked in your first reply.

Currently I'm stuck at this because I'm unable to verify that it's working - the demo gnuradio graphs [(here)](https://github.com/mossmann/pseudo-doppler/blob/master/df-mic-demo.grc) aren't working properly - either because they're out of date or because I'm missing some other part of the setup.

mossmann, 2021-02-14:
If I remember correctly, @docprofsky did not complete work on the hackrf host code for the sctimer demo. We were controlling it using some hacked up GreatFET host code, perhaps this branch: https://github.com/docprofsky/GreatFET-experimental/tree/hackrf-greatfet

straithe, 2021-03-20:
I'm going to close this as there hasn't been a response in a while. Please let us know if you still need assistance!

closed 2021-03-20 by straithe