greatscottgadgets/hackrf #744: osmosdr sink block not working

chinthakaranawaka, 2020-06-02:
when I generate flow graphs I got following error

Generating: '/home/chinthaka/Desktop/default1.py'
Generate Error: (NameError("'gain_mode16' is not defined"), 'osmosdr.sink(\n    args="numchan=" + str(${nchan}) + " " + ${args}\n)\n% if context.get(\'num_mboards\')() > 0:\n% if context.get(\'clock_source0\')():\nself.${id}.set_clock_source(${clock_source0}, 0)\n% endif\n% if context.get(\'time_source0\')():\nself.${id}.set_time_source(${time_source0}, 0)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 1:\n% if context.get(\'clock_source1\')():\nself.${id}.set_clock_source(${clock_source1}, 1)\n% endif\n% if context.get(\'time_source1\')():\nself.${id}.set_time_source(${time_source1}, 1)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 2:\n% if context.get(\'clock_source2\')():\nself.${id}.set_clock_source(${clock_source2}, 2)\n% endif\n% if context.get(\'time_source2\')():\nself.${id}.set_time_source(${time_source2}, 2)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 3:\n% if context.get(\'clock_source3\')():\nself.${id}.set_clock_source(${clock_source3}, 3)\n% endif\n% if context.get(\'time_source3\')():\nself.${id}.set_time_source(${time_source3}, 3)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 4:\n% if context.get(\'clock_source4\')():\nself.${id}.set_clock_source(${clock_source4}, 4)\n% endif\n% if context.get(\'time_source4\')():\nself.${id}.set_time_source(${time_source4}, 4)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 5:\n% if context.get(\'clock_source5\')():\nself.${id}.set_clock_source(${clock_source5}, 5)\n% endif\n% if context.get(\'time_source5\')():\nself.${id}.set_time_source(${time_source5}, 5)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 6:\n% if context.get(\'clock_source6\')():\nself.${id}.set_clock_source(${clock_source6}, 6)\n% endif\n% if context.get(\'time_source6\')():\nself.${id}.set_time_source(${time_source6}, 6)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 7:\n% if context.get(\'clock_source7\')():\nself.${id}.set_clock_source(${clock_source7}, 7)\n% endif\n% if context.get(\'time_source7\')():\nself.${id}.set_time_source(${time_source7}, 7)\n% endif\n% endif\n% if sync == \'sync\':\nself.${id}.set_time_unknown_pps(osmosdr.time_spec_t())\n% elif sync == \'pc_clock\':\nself.${id}.set_time_now(osmosdr.time_spec_t(time.time()), osmosdr.ALL_MBOARDS)\n% endif\nself.${id}.set_sample_rate(${sample_rate})\n% if context.get(\'nchan\')() > 0:\nself.${id}.set_center_freq(${freq0}, 0)\nself.${id}.set_freq_corr(${corr0}, 0)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode0}, 0)\nself.${id}.set_iq_balance_mode(${iq_balance_mode0}, 0)\nself.${id}.set_gain_mode(${gain_mode0}, 0)\n% endif\nself.${id}.set_gain(${gain0}, 0)\nself.${id}.set_if_gain(${if_gain0}, 0)\nself.${id}.set_bb_gain(${bb_gain0}, 0)\nself.${id}.set_antenna(${ant0}, 0)\nself.${id}.set_bandwidth(${bw0}, 0)\n% endif\n% if context.get(\'nchan\')() > 1:\nself.${id}.set_center_freq(${freq1}, 1)\nself.${id}.set_freq_corr(${corr1}, 1)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode1}, 1)\nself.${id}.set_iq_balance_mode(${iq_balance_mode1}, 1)\nself.${id}.set_gain_mode(${gain_mode1}, 1)\n% endif\nself.${id}.set_gain(${gain1}, 1)\nself.${id}.set_if_gain(${if_gain1}, 1)\nself.${id}.set_bb_gain(${bb_gain1}, 1)\nself.${id}.set_antenna(${ant1}, 1)\nself.${id}.set_bandwidth(${bw1}, 1)\n% endif\n% if context.get(\'nchan\')() > 2:\nself.${id}.set_center_freq(${freq2}, 2)\nself.${id}.set_freq_corr(${corr2}, 2)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode2}, 2)\nself.${id}.set_iq_balance_mode(${iq_balance_mode2}, 2)\nself.${id}.set_gain_mode(${gain_mode2}, 2)\n% endif\nself.${id}.set_gain(${gain2}, 2)\nself.${id}.set_if_gain(${if_gain2}, 2)\nself.${id}.set_bb_gain(${bb_gain2}, 2)\nself.${id}.set_antenna(${ant2}, 2)\nself.${id}.set_bandwidth(${bw2}, 2)\n% endif\n% if context.get(\'nchan\')() > 3:\nself.${id}.set_center_freq(${freq3}, 3)\nself.${id}.set_freq_corr(${corr3}, 3)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode3}, 3)\nself.${id}.set_iq_balance_mode(${iq_balance_mode3}, 3)\nself.${id}.set_gain_mode(${gain_mode3}, 3)\n% endif\nself.${id}.set_gain(${gain3}, 3)\nself.${id}.set_if_gain(${if_gain3}, 3)\nself.${id}.set_bb_gain(${bb_gain3}, 3)\nself.${id}.set_antenna(${ant3}, 3)\nself.${id}.set_bandwidth(${bw3}, 3)\n% endif\n% if context.get(\'nchan\')() > 4:\nself.${id}.set_center_freq(${freq4}, 4)\nself.${id}.set_freq_corr(${corr4}, 4)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode4}, 4)\nself.${id}.set_iq_balance_mode(${iq_balance_mode4}, 4)\nself.${id}.set_gain_mode(${gain_mode4}, 4)\n% endif\nself.${id}.set_gain(${gain4}, 4)\nself.${id}.set_if_gain(${if_gain4}, 4)\nself.${id}.set_bb_gain(${bb_gain4}, 4)\nself.${id}.set_antenna(${ant4}, 4)\nself.${id}.set_bandwidth(${bw4}, 4)\n% endif\n% if context.get(\'nchan\')() > 5:\nself.${id}.set_center_freq(${freq5}, 5)\nself.${id}.set_freq_corr(${corr5}, 5)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode5}, 5)\nself.${id}.set_iq_balance_mode(${iq_balance_mode5}, 5)\nself.${id}.set_gain_mode(${gain_mode5}, 5)\n% endif\nself.${id}.set_gain(${gain5}, 5)\nself.${id}.set_if_gain(${if_gain5}, 5)\nself.${id}.set_bb_gain(${bb_gain5}, 5)\nself.${id}.set_antenna(${ant5}, 5)\nself.${id}.set_bandwidth(${bw5}, 5)\n% endif\n% if context.get(\'nchan\')() > 6:\nself.${id}.set_center_freq(${freq6}, 6)\nself.${id}.set_freq_corr(${corr6}, 6)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode6}, 6)\nself.${id}.set_iq_balance_mode(${iq_balance_mode6}, 6)\nself.${id}.set_gain_mode(${gain_mode6}, 6)\n% endif\nself.${id}.set_gain(${gain6}, 6)\nself.${id}.set_if_gain(${if_gain6}, 6)\nself.${id}.set_bb_gain(${bb_gain6}, 6)\nself.${id}.set_antenna(${ant6}, 6)\nself.${id}.set_bandwidth(${bw6}, 6)\n% endif\n% if context.get(\'nchan\')() > 7:\nself.${id}.set_center_freq(${freq7}, 7)\nself.${id}.set_freq_corr(${corr7}, 7)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode7}, 7)\nself.${id}.set_iq_balance_mode(${iq_balance_mode7}, 7)\nself.${id}.set_gain_mode(${gain_mode7}, 7)\n% endif\nself.${id}.set_gain(${gain7}, 7)\nself.${id}.set_if_gain(${if_gain7}, 7)\nself.${id}.set_bb_gain(${bb_gain7}, 7)\nself.${id}.set_antenna(${ant7}, 7)\nself.${id}.set_bandwidth(${bw7}, 7)\n% endif\n% if context.get(\'nchan\')() > 8:\nself.${id}.set_center_freq(${freq8}, 8)\nself.${id}.set_freq_corr(${corr8}, 8)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode8}, 8)\nself.${id}.set_iq_balance_mode(${iq_balance_mode8}, 8)\nself.${id}.set_gain_mode(${gain_mode8}, 8)\n% endif\nself.${id}.set_gain(${gain8}, 8)\nself.${id}.set_if_gain(${if_gain8}, 8)\nself.${id}.set_bb_gain(${bb_gain8}, 8)\nself.${id}.set_antenna(${ant8}, 8)\nself.${id}.set_bandwidth(${bw8}, 8)\n% endif\n% if context.get(\'nchan\')() > 9:\nself.${id}.set_center_freq(${freq9}, 9)\nself.${id}.set_freq_corr(${corr9}, 9)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode9}, 9)\nself.${id}.set_iq_balance_mode(${iq_balance_mode9}, 9)\nself.${id}.set_gain_mode(${gain_mode9}, 9)\n% endif\nself.${id}.set_gain(${gain9}, 9)\nself.${id}.set_if_gain(${if_gain9}, 9)\nself.${id}.set_bb_gain(${bb_gain9}, 9)\nself.${id}.set_antenna(${ant9}, 9)\nself.${id}.set_bandwidth(${bw9}, 9)\n% endif\n% if context.get(\'nchan\')() > 10:\nself.${id}.set_center_freq(${freq10}, 10)\nself.${id}.set_freq_corr(${corr10}, 10)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode10}, 10)\nself.${id}.set_iq_balance_mode(${iq_balance_mode10}, 10)\nself.${id}.set_gain_mode(${gain_mode10}, 10)\n% endif\nself.${id}.set_gain(${gain10}, 10)\nself.${id}.set_if_gain(${if_gain10}, 10)\nself.${id}.set_bb_gain(${bb_gain10}, 10)\nself.${id}.set_antenna(${ant10}, 10)\nself.${id}.set_bandwidth(${bw10}, 10)\n% endif\n% if context.get(\'nchan\')() > 11:\nself.${id}.set_center_freq(${freq11}, 11)\nself.${id}.set_freq_corr(${corr11}, 11)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode11}, 11)\nself.${id}.set_iq_balance_mode(${iq_balance_mode11}, 11)\nself.${id}.set_gain_mode(${gain_mode11}, 11)\n% [... cut ...] nter_freq(${freq31}, 31)\nself.${id}.set_freq_corr(${corr31}, 31)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode31}, 31)\nself.${id}.set_iq_balance_mode(${iq_balance_mode31}, 31)\nself.${id}.set_gain_mode(${gain_mode31}, 31)\n% endif\nself.${id}.set_gain(${gain31}, 31)\nself.${id}.set_if_gain(${if_gain31}, 31)\nself.${id}.set_bb_gain(${bb_gain31}, 31)\nself.${id}.set_antenna(${ant31}, 31)\nself.${id}.set_bandwidth(${bw31}, 31)\n% endif\n')
>>> Failure

miek, 2020-07-28:
It looks like this is a bug in gr-osmosdr, I've submitted a patch to their mailing list but here it is in the meantime: https://github.com/miek/gr-osmosdr/commit/b1cf70f97c15f4d5ae206b5546a10a8d0e58a57e

I'll leave this open for now to keep track, until it gets closed upstream

miek, 2020-08-03:
Fixed in gr-osmosdr v0.2.1

closed 2020-08-03 by miek