greatscottgadgets/hackrf #746: Unable to install hackrf

titou420, 2020-06-15:
### Steps to reproduce
$ git clone https://github.com/mossmann/hackrf.git
$ cd hackrf/host
$ mkdir build
$ cd build
$ cmake ..

### Actual behaviour
Error occured and I can go further

### Version information
**Operating system**:
Operating System: BunsenLabs GNU/Linux 9.9 (Helium)
Kernel: Linux 4.9.0-11-amd64
Architecture: x86-64

### Output
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
-- Could NOT find LIBUSB (missing:  LIBUSB_LIBRARIES LIBUSB_INCLUDE_DIR) 
-- Setting udev rule group to - plugdev
-- HackRF udev rules will be installed to '/etc/udev/rules.d' upon running 'make install'
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBUSB_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf/src
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf/src
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf/src
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf/src
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf/src
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf/src
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf/src
   used as include directory in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf/src
LIBUSB_LIBRARIES (ADVANCED)
    linked by target "hackrf" in directory /home/local/ADDDOM/t.quelen/hackrf/host/libhackrf/src

-- Configuring incomplete, errors occurred!
See also "/home/local/ADDDOM/t.quelen/hackrf/host/build/CMakeFiles/CMakeOutput.log".
See also "/home/local/ADDDOM/t.quelen/hackrf/host/build/CMakeFiles/CMakeError.log".

GravesJake, 2020-06-15:
It looks like you're missing pkg-config and libusb, try the following:
```
$ sudo apt-get install pkg-config
$ sudo apt install libusb-1.0-0
```

titou420, 2020-06-16:
Yesterday I already try to install these package.
For libusb I got : libusb-1.0-0 is already the newest version (2:1.0.21-1).
For pkg-config I got this :
The following packages were automatically installed and are no longer required:
  libmicrodns0 libv8-3.14.5 linux-image-4.9.0-9-amd64
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  pkg-config
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 63.3 kB of archives.
After this operation, 193 kB of additional disk space will be used.
Get:1 http://ftp.fr.debian.org/debian stretch/main amd64 pkg-config amd64 0.29-4+b1 [63.3 kB]
Fetched 63.3 kB in 0s (252 kB/s)    
Traceback (most recent call last):
  File "/usr/bin/apt-listchanges", line 29, in <module>
    import apt_pkg
ModuleNotFoundError: No module named 'apt_pkg'
Selecting previously unselected package pkg-config.
(Reading database ... 308520 files and directories currently installed.)
Preparing to unpack .../pkg-config_0.29-4+b1_amd64.deb ...
Unpacking pkg-config (0.29-4+b1) ...
Setting up pkg-config (0.29-4+b1) ...
Processing triggers for man-db (2.7.6.1-2) ...

And I can't install the hackrf's soft ....
I already install it on a Debian computer and Fedora, with no problem

titou420, 2020-06-16:
Mmmmh, finally succeed, I have juste install the lib-usb-1.0-0-dev package ..
It's weird that I've lib-usb regular package and it doesn't work ..
Thanks :)

GravesJake, 2020-06-16:
Oh right, I forgot there is a weirdness with the libusb version. Glad you got it figured out!

closed 2020-06-16 by GravesJake

miek, 2020-06-16:
Just for future reference: Debian/Ubuntu split out headers (and other stuff required for compiling against a library) into a separate package with the `-dev` suffix, so that's why `libusb-1.0-0-dev` was required