greatscottgadgets/hackrf #747: starting from new

mordevious, 2020-06-15:
I am just starting with Linux ubuntu and have tried to find what steps I have to follow to get gnuradio, osmocom and hackrf working .  I have tried to find advice on websites but they seem to offer different solutions that my version of ubuntu (20.04 lts) doesn't have also a lot of the links to help are missing.  I have reloaded Ubuntu to a clean drive and would appreciate any help in the steps I need to take to get my Hackrf working.
I have tried to used the win 10 version of gnu radio but can't get if to work with hackrf whilst other sdr programs seem to have no problems.
Any help and advice would be appreciated

eried, 2020-06-15:
What does not work? Describe are your steps

mordevious, 2020-06-15:
I can get gnu radio using ubuntu search but it does not have osmocom available. I have only been using Ubuntu for a couple of days and as I tried to follow the advice and sequences on various site, e.g both main osmocom and hackrf sites I seem to get lost and get told that the command words are not available.  Then I saw that you had to load the software in a certain order but no indication what that was.  Please assume I am a complete numpty with this as although I am an Rf engineering instructor using sdr and software is relatively new to me.  My apologies but I did not save any details before I wiped the hard drive and have just finished reloading Ubuntu and am about to do the initial setup

mordevious, 2020-06-15:
If I can get it working I will publish the sequence for others to follow especially my work colleagues and happily credit any assistance

eried, 2020-06-15:
ok but why dont you start with simpler problem solving steps. Does the hackrf appear connected? does it has lights on?

mordevious, 2020-06-15:
yes

mordevious, 2020-06-15:
and it works with other software on Win 10 so I'm happy it's not a physical problem

mordevious, 2020-06-15:
sorry to be slow responding but I'm also supporting a web course while I'm doing this

eried, 2020-06-15:
nice, but then why you are asking here if the problem is not the hackrf?

mordevious, 2020-06-15:
because i don't know where to ask or what I'm doing with it. if you could advise me on the best place to go to get the software loading sequence I would appreciate it

GravesJake, 2020-06-15:
If you haven't checked already, there's installation information in the [Getting Started](https://github.com/mossmann/hackrf/wiki/Getting-Started-with-HackRF-and-GNU-Radio#software-setup) page as well as the [Operation System Tips](https://github.com/mossmann/hackrf/wiki/Operating-System-Tips) in the [hackrf wiki](https://github.com/mossmann/hackrf/wiki).

Have you tried using the following commands to install osmocom and hackrf tools on your system?
```
$ sudo apt-get install gr-osmosdr
$ sudo apt-get install libhackrf
```

mordevious, 2020-06-17:
thanks for the help it is much appreciated.

GravesJake, 2020-06-17:
No problem!

closed 2020-06-17 by GravesJake