greatscottgadgets/hackrf #749: Every sample rate change requires reset

brian-armstrong, 2020-06-17:
### Steps to reproduce
1. Set a sample rate on the HackRF One. This can be via either Windows or Linux, and a variety of applications (SDR#, Soapy SDR, libhackrf)
2. Attempt to set a new sample rate. This will fail.
3. Reset the HackRF One via reset button or cable in/out
4. Set a sample rate. This will succeed.

### Expected behaviour
The sample rate should be changeable without resetting the device

### Actual behaviour
The sample rate cannot be changed without resetting the device

### Version information
**Operating system**:
Windows 10
Ubuntu 18.04

**hackrf_info output:**
hackrf_info version: git-43e6f99
libhackrf version: git-43e6f99 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc317b8207
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x004f4f6f

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
SDR# v1.0.0.1732

rtl_433 version 20.02-65-g02a009c branch master at 202006142311 inputs file rtl_tcp RTL-SDR SoapySDR

dump1090 fork

### Output
```
Insert any commandline or build output here
```
rtl_433: "WARNING: sync read failed. -1"
dump1090: "No data received from the SDR for a long time, it may have wedged"

miek, 2020-06-17:
You've got a mismatch between host & firmware versions which I think is causing this. You should either install the host tools from the 2018.01.1 release, or build the firmware from master & program both the flash & CPLD.

brian-armstrong, 2020-06-17:
Thanks for the quick reply.

By host version do you mean the drivers in use? I don't think I know what I should be updating specifically. I assume on Windows that means the USB driver although I'm not sure where to find that, I believe I'm just using one that Windows found automatically. I could also imagine you mean the libhackrf version compiled into SDR#?

On linux I'm guessing this is libhackrf that I need to update?

miek, 2020-06-17:
Yep, it's the version of libhackrf & also the hackrf command-line tools (hackrf_info, hackrf_transfer, etc.). There's no extra driver to update on Windows - all the device-specific stuff happens in libhackrf.

Though the problem is the other way around to what you're saying: the hackrf_info output you posted above shows that you already have the latest libhackrf, it's the firmware that's older.

brian-armstrong, 2020-06-17:
Ok, I went ahead and updated on Linux but I'm seeing the same issue persist. Here's the new hackrf_info output. It claims the version of libhackrf is "unknown (0.5)" but when I installed from the release, it claims to have installed 0.5, so I suspect this is the version from the release

$ hackrf_info
hackrf_info version: 2018.01.1
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc317b8207
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x004f4f6f

brian-armstrong, 2020-06-19:
Is there anything else I can try to diagnose or debug this? It's a bit annoying to get up and poke the reset button each time I want to try a different mode or change the sample rate.

miek, 2020-06-19:
How did you install the version you had originally (git-43e6f99) and how did you install version 2018.01.1? I wonder whether you have ended up with both versions installed and some tools may still be using the original one.

brian-armstrong, 2020-06-20:
The git version wasn't installed. I built it and used the build artifacts to get the info, but I didn't install it. Prior to that I had just used apt install libhackrf-dev on Ubuntu 18.

With the current version, I copied the release zip, unzipped it, then make/make install

I can try apt uninstall so that it has to be the version from the zip

brian-armstrong, 2020-06-20:
Ok so yeah, I now have a clean looking hackrf_info, though the problem continues. I also tried removing a USB extension cable in the chain and that didn't help either.

```
$ ./hackrf_info hackrf_info version: 2018.01.1
libhackrf version: 2018.01.1 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc317b8207
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x004f4f6f
```

brian-armstrong, 2020-06-21:
Has anyone else encountered this issue before or know if it's common? I'm wondering if I actually just have a defective unit. If sending it back to Amazon is all it takes to fix it, that seems easy enough.

eried, 2020-06-21:
I tried in SDR# in Windows and it does not require reset. Have you tried another usb cable? I experienced some disconnects with cheap ones

brian-armstrong, 2020-06-21:
Yeah, I tried a completely different cable from the one it came with. Same results.

brian-armstrong, 2020-06-22:
It looks like the window for my return closes relatively soon. I'd like to try to avoid returning if I can but I can't think of other mitigations to try and I want to make sure I return while that's still an option. Is there anything else I might be able to try to fix this?

miek, 2020-06-24:
I was able to reproduce it (I think) using the set of host/firmware version in the first post. Other than that, I've not encountered this before and I don't think it's common.
It would be helpful to see if you can reproduce it with just the hackrf-tools:

    hackrf_transfer -r /dev/null -s 20e6
    hackrf_transfer -r /dev/null -s 10e6
    etc.

Doing that a few times eventually hung and required a HackRF reset, when I had mismatched versions.

brian-armstrong, 2020-06-26:
Interesting. That actually fails on the second attempt every time, and without changing the sample rate.
```
$ hackrf_transfer -r /dev/null -s 10e6
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
19.9 MiB / 1.001 sec = 19.9 MiB/second
19.9 MiB / 1.001 sec = 19.9 MiB/second
20.2 MiB / 1.000 sec = 20.2 MiB/second
19.9 MiB / 1.001 sec = 19.9 MiB/second
^CCaught signal 2
16.8 MiB / 0.838 sec = 20.0 MiB/second

Exiting...
Total time: 4.84182 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
```
```
$ hackrf_transfer -r /dev/null -s 10e6
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
 0.0 MiB / 1.000 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 1.00037 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
```

brian-armstrong, 2020-06-26:
Also, I tried some basic receive tests on my MacOS laptop, and I can't seem to reproduce the issue with the radio at all. I'm able to start and stop receiving with various programs, sample rates, etc. and things seem to work. Somehow my Windows PC and the Ubuntu VM inside it have the issue but my other computer does not.

brian-armstrong, 2020-06-26:
I'm also noticing this in the syslog in the Ubuntu VM
```
Jun 25 18:37:10 ubuntu upowerd[3456]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:11.0/0000:02:03.0/usb1/1-1
Jun 25 18:37:13 ubuntu upowerd[3456]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:11.0/0000:02:03.0/usb1/1-1/1-1:1.0
Jun 25 18:37:28 ubuntu upowerd[3456]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:11.0/0000:02:03.0/usb1/1-1/1-1:1.0
Jun 25 18:37:29 ubuntu upowerd[3456]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:11.0/0000:02:03.0/usb1/1-1/1-1:1.0
Jun 25 18:37:31 ubuntu upowerd[3456]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:11.0/0000:02:03.0/usb1/1-1/1-1:1.0
```

brian-armstrong, 2020-06-28:
Weirdly, I tried updating my Windows USB device driver with Zadig and then resetting it back to the Windows-supplied driver, and now I seem to be able to start and stop receive inside Windows without issue, but the Ubuntu guest VM still cannot stop and start without a reset. I am definitely suspecting that it's either VMWare's USB driver or libusb or similar in Ubuntu. It's not clear to me how a USB driver could cause this though as board is definitely still visible and can be queried with hackrf_info. It seems to be just putting it into RX mode that's semi-failing, and really it's just that retrieving samples times out forever.

How hard is it to get debug traces or logging from the firmware? I'm thinking it might be interesting to to try to probe out what state the board thinks it's in vs what libhackrf sees.

closed 2021-05-23 by straithe