greatscottgadgets/hackrf #750: Not able to install hackrf

aquablue07, 2020-06-17:
Steps to reproduce
$ git clone https://github.com/mossmann/hackrf.git
$ cd hackrf/host
$ mkdir build
$ cd build
$ cmake ..

Actual behaviour
FFTW error occurs

Version information
Operating system:
Operating System: Raspberry pi, Raspbian buster

Output
CMake Error at /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find FFTW (missing: FFTW_LIBRARIES FFTW_INCLUDES)
Call Stack (most recent call first):
  /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindFFTW.cmake:28 (find_package_handle_standard_args)
  hackrf-tools/src/CMakeLists.txt:26 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/hackrf/host/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/hackrf/host/build/CMakeFiles/CMakeError.log".

miek, 2020-06-17:
You need to install the prerequisites listed here: https://github.com/mossmann/hackrf/tree/master/host#prerequisites-for-linux-debianubuntu - libfftw3-dev in particular for this error, but you'll probably need some of the others too.

aquablue07, 2020-06-17:
my bad...was an oversight! Thank you!

closed 2020-06-17 by aquablue07