greatscottgadgets/hackrf #751: Jump in noise floor of several decibels

jbaggs, 2020-06-21:
I have been experiencing an issue with the noise floor suddenly increasing by 4-10 dB (depending on amplification) over a wide portion of spectrum (peak at least 30 mHz wide) for some time now. It can last anywhere from a number of minutes to hours, independent of time of day, and seemingly independent of temperature. It does not appear to matter how long the HackRF has been in use, and is not resolved with a restart. It is not affected by whether or not the computer running it is on wall power or battery. 

I would lean towards outside interference, except for the fact that it is so broad, and not fitting any patterns I can discern. Is there any component that could be failing that might cause the issue?

Attached are images of a 5 mHz section of VHF spectrum, with and without the noise, both with AMP on, and off.
[image]
[image]
[image]

Thanks in advance for any ideas.
[image]

mossmann, 2020-06-21:
Have you observed this in other bands? Does it happen with alternative USB cables?

jbaggs, 2020-06-21:
I see it on low VHF as well. (e.g. 41.7 - 47.7 mHz) I can try some other bands, but will have to wait awhile to get both a clean and high noise picture. Switching USB cables doesn't help it. The one I normally use with the HackRF has ferrite chokes. Some others I have add noise on top of the reported problem.

Videogamer555, 2020-06-22:
I'm guessing also outside interference. Possibly some ultrawideband communication mode of some kind.

jbaggs, 2020-06-22:
This is at least 160 MHz wide at a constant power level with no signs of intentional modulation. It's definitely not an intentional emission, if it is external.

jbaggs, 2020-06-22:
I put an AM demodulator in the middle of the spectrum where it seemed the worst, and noted there is a roughly 120 Hz raspy tone in with the the normal background noise. I had expected if there was line noise I would be hearing something more around 60 Hz, but did some reading and found that high voltage lines often arc on both sides of the cycle. I also threw together an impromptu mobile antenna and drove around a bit, confirming it is location dependent. Glad it is not the device.

closed 2020-06-22 by jbaggs