greatscottgadgets/hackrf #753: Hack rf stops during operation

aquablue07, 2020-06-22:
### Steps to reproduce
1.hackrf_sweep -f 2400:2500 -r new.csv
2.
3.

### Expected behaviour
Tell us what you expect should happen
302 total sweeps completed, 81.26 sweeps/second
305 total sweeps completed, 80.97 sweeps/second
310 total sweeps completed, 81.22 sweeps/second
313 total sweeps completed, 80.94 sweeps/second
318 total sweeps completed, 81.18 sweeps/second
321 total sweeps completed, 80.92 sweeps/second
326 total sweeps completed, 81.15 sweeps/second
329 total sweeps completed, 80.89 sweeps/second
334 total sweeps completed, 81.12 sweeps/second
337 total sweeps completed, 80.86 sweeps/second
342 total sweeps completed, 81.09 sweeps/second
347 total sweeps completed, 81.31 sweeps/second
350 total sweeps completed, 81.06 sweeps/second
Caught signal 2
353 total sweeps completed, 81.03 sweeps/second

Exiting...
Total sweeps: 353 in 4.35667 seconds (81.03 sweeps/second)
^Chackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

### Actual behaviour
Tell us what happens instead
"Couldn't transfer any data for one second."

hackrf_sweep stops occasionally and the above message appears

### Version information
**Operating system**: Raspbian buster

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 2400 MHz to 2500 MHz
Stop with Ctrl-C
2 total sweeps completed, 39.94 sweeps/second
7 total sweeps completed, 69.87 sweeps/second
10 total sweeps completed, 66.54 sweeps/second
15 total sweeps completed, 74.85 sweeps/second
18 total sweeps completed, 71.85 sweeps/second
18 total sweeps completed, 59.88 sweeps/second

Couldn't transfer any data for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 18 in 0.30072 seconds (59.88 sweeps/second)
Insert any commandline or build output here
```

eried, 2020-06-22:
Sounds like a crappy usb cable, can you get a super short/thick 15/30 cm one?

aquablue07, 2020-06-23:
> Sounds like a crappy usb cable, can you get a super short/thick 15/30 cm one?
I tried using a shorter and thicker cable. now, it says 
"hackrf_open() failed: Resource busy (-1000)"

straithe, 2021-02-25:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-02-25 by straithe

metayan, 2021-03-18:
This occured thrice here now after upgrading to https://github.com/mossmann/hackrf/commit/6fbc6dfea4e2cf59ffb8269bc0d030ad759d7a11 , i.e. after the "Disable USB interrupts during sweep set_freq call" fix in https://github.com/mossmann/hackrf/pull/847
```
15 total sweeps completed, 1.23 sweeps/second 
15 total sweeps completed, 1.22 sweeps/second 
15 total sweeps completed, 1.22 sweeps/second 
                                                                    
Couldn't transfer any data for one second.    
                                                                    
Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 15 in 12.32206 seconds (1.22 sweeps/second)
hackrf_stop_rx() done                         
hackrf_close() done                           
hackrf_exit() done                            
exit                                          
                                                                    
real    0m13.348s                             
user    0m0.506s                              
sys     0m0.116s
```
```
847 total sweeps completed, 1.30 sweeps/second
847 total sweeps completed, 1.30 sweeps/second
847 total sweeps completed, 1.30 sweeps/second

Couldn't transfer any data for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 847 in 649.92969 seconds (1.30 sweeps/second)
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
exit

real    10m50.597s
user    0m53.864s
sys     0m14.475s
```
```
2485 total sweeps completed, 1.30 sweeps/second
2485 total sweeps completed, 1.30 sweeps/second
2485 total sweeps completed, 1.30 sweeps/second

Couldn't transfer any data for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 2485 in 1906.01221 seconds (1.30 sweeps/second)
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
exit

real    31m46.775s
user    2m16.818s
sys     0m29.613s
```
The last one was piped to `/dev/null`.

```
hackrf_info version: git-6fbc6df
libhackrf version: git-6fbc6df (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc2919845f
Board ID Number: 2 (HackRF One)
Firmware Version: git-6fbc6df (API:1.04)
Part ID Number: 0xa000cb3c 0x00644f56
```
Invoked with
```
time nice hackrf_sweep -f 0:6240 -w 200000 > /dev/null
```

#### with old version ####
Giving version https://github.com/mossmann/hackrf/commit/f2021608a6a0a860624c7b517568632ce601d2af a try now.
```
hackrf_info version: git-6fbc6df               
libhackrf version: git-6fbc6df (0.5)           
Found HackRF                                   
Index: 0                                       
Serial number: 000000000000000087c867dc2919845f
Board ID Number: 2 (HackRF One)                
Firmware Version: git-f202160 (API:1.04)       
Part ID Number: 0xa000cb3c 0x00644f56
```
Better, but also failed.
```
17334 total sweeps completed, 1.30 sweeps/second
17334 total sweeps completed, 1.30 sweeps/second
17334 total sweeps completed, 1.30 sweeps/second

Couldn't transfer any data for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 17334 in 13291.89551 seconds (1.30 sweeps/second)
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
exit

real    221m32.461s
user    16m4.466s
sys     3m30.422s
```
This last one was also piped to `/dev/null`.

Running on a server
`Linux aum 4.15.0-33-generic #36-Ubuntu SMP Wed Aug 15 16:00:05 UTC 2018
 x86_64 Intel(R) Core(TM) i5-3210M CPU @ 2.50GHz GenuineIntel GNU/Linux`
 with not much load
`22:53:27 up 925 days, 20:39,  5 users,  load average: 0.13, 0.11, 0.06`

Any ideas how to test more are welcome.
The second part of the original issue, where nothing is transmitted at all, is not seen here.

miek, 2021-03-18:
We also just pushed up a fix for how that timeout is calculated (#853), could you give that a try?

metayan, 2021-03-19:
Certainly.

metayan, 2021-03-19:
What sweep parameters would test it the best?
Running
`hackrf_sweep -f 0:7250 -r /dev/null`
gave
```
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 7260 MHz
Stop with Ctrl-C
1 total sweeps completed, 1.00 sweeps/second
2 total sweeps completed, 1.00 sweeps/second
```
```
  PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
18088  0.0  0.1 106284  4152 pts/12   Sl+  00:30   0:00 hackrf_sweep -f 0 7250 -r /dev/null
18088  0.0  0.1 106284  4152 pts/12   Sl+  00:30   0:00 hackrf_sweep -f 0 7250 -r /dev/null
18088  3.9  0.1 106284  4152 pts/12   Sl+  00:30   0:34 hackrf_sweep -f 0 7250 -r /dev/null
```
```
1123 total sweeps completed, 1.12 sweeps/second
1124 total sweeps completed, 1.12 sweeps/second
^CCaught signal 2

Exiting...
Total sweeps: 1124 in 1003.40637 seconds (1.12 sweeps/second)
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose() done
exit
```

A second run with parameters as in https://github.com/mossmann/hackrf/issues/753#issuecomment-802388392
`time nice hackrf_sweep -f 0:6240 -w 200000 > /dev/null`
```
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 6240 MHz
Stop with Ctrl-C
1 total sweeps completed, 1.00 sweeps/second
2 total sweeps completed, 1.00 sweeps/second
```
```
  PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
25172  0.0  0.1 106416  6404 pts/12   SNl+ 00:50   0:00 hackrf_sweep -f 0 6240 -w 200000
25172  0.0  0.1 106416  6404 pts/12   SNl+ 00:50   0:00 hackrf_sweep -f 0 6240 -w 200000
25172  8.3  0.1 106416  6404 pts/12   SNl+ 00:50   0:08 hackrf_sweep -f 0 6240 -w 200000
```
It's at
```
822 total sweeps completed, 1.30 sweeps/second
823 total sweeps completed, 1.30 sweeps/second
```
now and seems stable so far.
I'll let it run for long, unless you suggest to use different parameters.

metayan, 2021-03-19:
Yeah, it seems to be good now:
```
49012 total sweeps completed, 1.30 sweeps/second
49013 total sweeps completed, 1.30 sweeps/second
```
Great.