greatscottgadgets/hackrf #754: large pulses/spikes in signal

spiantino, 2020-06-28:
I'm seeing big, whole-spectrum power spikes in cubicsdr. Here's a screenshot from Ubuntu:

https://imgur.com/a/RWIYnKx

The funny thing is I've tried multiple antennas, different computers and OS/s and these pulsing signals are present on any signal (though much more prevalent above 1.5ghz). I'd blame some kind of local interference, but I don't see them on E4000 based SDRs and am able to decode GOES and Inmarsat in that band pretty easily.

Could my HackRF One have an issue?

`
serkan@sp-linux:~/gr-iridium$ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc31419b07
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x007d4365
`

GravesJake, 2020-07-15:
Might be unrelated but your host tools don't match your Firmware version.

Can you try downloading the [Release Package](https://github.com/mossmann/hackrf/releases/tag/v2018.01.1) and [updating your host tools](https://github.com/mossmann/hackrf/tree/master/host#how-to-build-the-host-software-on-linux) to match your Firmware Version (2018.01.1).

miek, 2020-07-23:
I suspect these signals are images of strong wifi signals. Reducing your gains might help ("AMP", in particular, should be off in most cases), but a band-pass filter for your frequency of interest would be best.

closed 2020-08-18 by miek