greatscottgadgets/hackrf #755: More Question Than Issue

rogueops, 2020-07-03:
I have a version of custom firmware on my HackRF that I cannot get again as the project died so I was wondering using either the dfu tools or the hackrf tools, is it possible to copy/download the current firmware from my HackRF before I try new firmware? Is it possible to backup the current firmware? Thanks for anyhelp in advance!

miek, 2020-07-03:
You should be able to read it back using `hackrf_spiflash -r backup.bin`. If possible, I'd try flashing it to a second HackRF to test it out before erasing the first.

rogueops, 2020-07-03:
Thats what I thought after looking at the man/help, but I wanted to be
double sure, thanks for the reply

closed 2020-07-03 by rogueops