greatscottgadgets/hackrf #758: Connecting a HackRF to a Digilent PMOD GPS

JAkin-at-ASSET, 2020-07-06:
I am seeking advice regarding the best way to directly connect a HackRF to a Digilent PMOD GPS Receiver.  The PMOD GPS features a GlobalTop FGPMMOPA6H GPS antenna module that utilizes the MediaTek GPS MT3329.

Is it sufficient to connect the two devices via simple SMA cable?

miek, 2020-07-23:
I think you've probably got this answered on the mailing list now, but I'll give an answer anyway. You generally shouldn't connect directly from the HackRF to a GPS receiver. GPS receivers expect a signal level many orders of magnitude lower than the HackRF outputs, so you need a lot of attenuation inline to avoid damage & to get correct results.

closed 2020-07-23 by miek

JAkin-at-ASSET, 2020-07-27:
Thank you for your reply, Mike.

Can you offer guidance, or direct me to resources regarding the proper type, or strength of anteaters I should use?

Also, I watched several demonstration videos in which people were able to detect the GPS transmission using their cell phones.  The HackRF’s transmit LED is definitely on, but my phone and any other GPS devices I have used to try to detect the signal don’t appear to be doing so.  Could you offer any suggestions on what I might try?

Thanks again.