greatscottgadgets/hackrf #759: HackRf one not working..

Fabtolab, 2020-07-08:
### Steps to reproduce
1. Turning on..
2. Connecting PC to HackRF using the provided cable
3. Turning on SDR sharp in PC 

### Expected behaviour-
i wanna play fm radio using sdr sharp and gqrx.

### Actual behaviour
it is not  playing fm at any frequency and it is not showing any fm frequency in both sdr sharp and gqrx.

### operating system
windows  10,ubuntu 20.04,kali linux 2020.4

Gqrx version-2.11.5
Sdr sharp version-x86 rev 1732

### Additional Comments
Also it is not capturing car key fob signals
I also tried same things with my friend's hackrf his hackrf is working fine playing radio and capturing and transmitting car key fobs.

[image]

GravesJake, 2020-07-08:
Can you try downloading the [Release Package](https://github.com/mossmann/hackrf/releases/tag/v2018.01.1) and [updating your host tools](https://github.com/mossmann/hackrf/tree/master/host#how-to-build-the-host-software-on-linux) to match your Firmware Version (2018.01.1). 

After that can you post the output of `lsusb` and `hackrf_info` please?

GravesJake, 2020-07-08:
Also screenshots of your fm capture would be helpful.

Fabtolab, 2020-07-11:
Done but still no success..
Let me know..

[image]
[image]
[image]
[image]

GravesJake, 2020-07-13:
What do the settings on the right side of the gqrx window look like? Also it doesn't appear that the host tools were properly updated to match the firmware version (hackrf_info version: unknown; libhackrf version: unknown)

Fabtolab, 2020-07-19:
Hi Graves,
Customer is not getting it to work even after the updates. He is comparing his unit with another previous unit he has and upon doing the same things he is not getting the frequencies. 
Also he is saying that his other hackrf purchased previously also had hacked info and libhackrf versions unknown..
Please suggest..
Thanks,
Niraj, Fab.to.Lab

GravesJake, 2020-07-21:
Could be a hardware issue, I'll email you with RMA info.

closed 2020-07-21 by GravesJake

Fabtolab, 2020-07-22:
Thanks Graves,
waiting..
We will place a new order soon. 
You can send the replacement alongwith.
We will share the shipping tracker asap.
Thanks,
Niraj
Fab.to.Lab