greatscottgadgets/hackrf #762: USB issue - any possibility to flash firmware not using USB port?

yann44, 2020-07-16:
Hi there,

Like most of issues in the forum, my hackrf doesn't communicate with my computer anymore.
When it is plugged, 3V3 led ON, 1V8 led ON, RF led ON, USB led OFF

4 different cables, same issue, 3 different computers (Win10, MacOs, Kali) same issue.

I would like to install last firmware but no communication between computers and hackrf, even in DFU mode.

Is there any possibility to flash it using wires on P20, or P22 or P28, or anywhere else?

hackrf_info

hackrf_info version: unknown
libhackrf version: unknown (0.5)
No HackRF boards found.

thanks a lot for your help!

Yann

NoahJaehnert, 2020-07-21:
I am having the exact same issue with my HackRF Blue. I detailed my issues in the thread on the hackrf-dev mailing list (https://pairlist9.pair.net/pipermail/hackrf-dev/2020-May/002924.html), and a few folks tried to help, but still cannot get my HackRF blue to respond via USB at all, not in regular nor DFU mode.

miek, 2020-07-23:
There's no simple way of re-flashing it (even using jtag, the actual flash programming happens over usb with `hackrf_spiflash`).
This sounds like a hardware failure in either the USB connector or ESD protection behind it. If you got your HackRF from one of our [resellers](https://greatscottgadgets.com/wheretobuy/) we can sort out a replacement.

vasja2000, 2020-08-09:
у меня уже год hackrf  и portapack лежат, такая же проблема.

closed 2020-08-18 by miek