greatscottgadgets/hackrf #765: The hackrf_transfer does not support large file (>2Gb) in rasberry pi

jxltom, 2020-07-24:
### Steps to reproduce
1. ```hackrf_transfer -t some_large_file.bin```

### Expected behaviour
It should transfer.

### Actual behaviour
```Failed to open file some_large_file.bin```

### Version information
**Operating system**:
raspeberry pi

**hackrf_info output:**
```
hackrf_info version: git-43e6f99*
libhackrf version: git-43e6f99* (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000026b468dc2e2d678f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x007d4773
```

### Output
```
pi@raspberrypi:~/hackrf/host/build $ sudo hackrf_transfer -t ~/Desktop/test.img -f 1575420000 -s 2600000 -a 1 -x 47
Failed to open file: /home/pi/Desktop/test.img
```

miek, 2020-07-24:
I assume you're running a 32bit OS here? Which model of Raspberry Pi is it? Which version of the OS?

This seems related to #152, but maybe that wasn't fixed quite correctly. Could you try editing `hackrf_transfer.c`, move the `#define _FILE_OFFSET_BITS 64` line to the very top of the file, rebuild, and test again?

mossmann, 2020-07-26:
@miek noticed that this appears to have been fixed in #716 which was just merged. Can you please pull and retest?

jxltom, 2020-07-30:
@miek @mossmann Thanks for the reply. I've pulled the latest code and tested in my rpi. I can confirm the issue is fixed. Thanks!

closed 2020-07-30 by jxltom