greatscottgadgets/hackrf #770: What is the transmission distance for HackRF and what is the gain of antenna ANT500

komalkumar6655, 2020-08-01:
#
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

miek, 2020-08-18:
It's not possible to give a number for transmission distance - it depends on way too many factors. The gain of the ANT500 is unspecified.

closed 2020-08-18 by miek

komalkumar6655, 2020-08-19:
Thank you for your response.
I just want to know what are the factors that the transmission distance
depends on.

*With regards,*
*S. Komal Kumar.*
*M.E. (Communication engineering), ECE*
*CBIT, Hyderabad.*