greatscottgadgets/hackrf #772: Hackrf sweep periodic tuning problem

blardyblar, 2020-08-07:
### Steps to reproduce
1. Run hackrfsweep in automated script (collects 30 sweeps in about 2 seconds (and averages them), repeated every minute)
2. Plot output over the course of many hours

### Expected behaviour
Sweeps same frequency range every time and returns valid data

### Actual behaviour
Periodically returns low level noise, as if it's not tuning at all (annotation A in screenshot, but it's pretty obvious). 
Occasionally returns completely bogus data, like it's tuning to something, but it's not right at all.  (annotation B in the screenshot)
Somehow fixes itself eventually and goes back to normal on it's own

### Version information
**Operating system**:
Debian linux, kernel 5.7.12

Hackrf tools and board firmware compiled from Github on 2020-08-04
(problem isn't specific to this release - this happened before, but there used to be a nasty USB bug that would panic the kernel running hackrf sweep automated (that's been fixed - THANK YOU for that!!), so it was harder to see, but I have seen the same behaviour before running sweeps by hand. I used to hit the reset button on the board and it would go back to normal, but now that I can run it automated, I see it somehow resolves itself sometimes, which is interesting.

**hackrf_info output:**
hackrf_info version: git-6e5cbda
libhackrf version: git-6e5cbda (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000321864c8386d781d
Board ID Number: 2 (HackRF One)
Firmware Version: git-6e5cbda (API:1.04)
Part ID Number: 0xa000cb3c 0x0068474c

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
see attached waterfall plot. This is about 18 hours of capture, each row of the waterfall is a minute. Capturing about 500MHz of spectrum. You can see the dropouts periodically ('A' in the screenshot), and the random garbage data ('B' in the screenshot). The return to normal behaviour is with no user intervention. Somehow it just starts to work again properly on it's own.
![hackrfsweep_bug](https://user-images.githubusercontent.com/15389119/89599996-cccb2680-d815-11ea-986b-bca11e670923.png)

```

dails08, 2020-08-10:
I get something similar. When running hackrf_sweep as a subprocess from a python script every five seconds, it will occasionally run the sweep function, but conduct no sweeps and return no data. Like you, it'll start working again on its own after a random amount of time.

blardyblar, 2020-08-10:
I added a simple detection to my script if the max-min value of the data returned is too small (it returned garbage data), I reboot the hackrf (I built a small board that can switch the USB power on and off using essentially an arduino and a couple mosfets). There is definitely a bug in the firmware that leads to this condition. Rebooting the HackRF board is an ugly kludge, but it works. Fixing the firmware bug would be way better!

miek, 2020-09-10:
@dails08:
> I get something similar. When running hackrf_sweep as a subprocess from a python script every five seconds, it will occasionally run the sweep function, but conduct no sweeps and return no data. Like you, it'll start working again on its own after a random amount of time.

Are you running the software/firmware from git master? If not, you might be hitting the issue that was fixed in #702 

@blardyblar:
> I added a simple detection to my script if the max-min value of the data returned is too small (it returned garbage data), I reboot the hackrf (I built a small board that can switch the USB power on [...]

I haven't any luck in reproducing this so far. If you're able to give a test case that I can run, I can take another look.

straithe, 2021-01-24:
Are you still having trouble? We will close this issue if we don't get a reply within the next month.

dails08, 2021-01-24:
I haven't been exercising it at all recently, so I don't have any updates.
Go ahead and close it, I say, and if it comes up again I'll wander on back.

blardyblar, 2021-01-24:
Yes, I still have the same issues. I don't have any special way to reproduce it other than run many short sweeps in a script, and eventually you won't get real data. The only fix is to hit the reset button on the hackrf, or unplug / replug the usb.
It seems like something gets left in an inconsistent state in the firmware on the device when the sweep is stopped, but quite rarely.

I also haven't updated the firmware or libhackrf since August when I opened this ticket, so if there's reason to believe this was fixed by some other update, I can try updating the firmware and libhackrf, but if nothing was changed there, nothing has changed with the behaviour of the device either.

straithe, 2021-01-26:
Thank you for the update. I've updated this to "bug" status and we'll take a deeper look when we can.

gozu42, 2021-03-13:
(i will spare you the week+ adventure of how i got here, two page rant available at request)
after crossing the snakepit at the bottom of this rabbit hole, i am now reasonably confident the issue here is the mixer.

in particular hackrf_close() seems to (sometimes) leave the mixer in a bad state when called while a sweep is running.

the most likely candidate seems to be the call to mixer_disable().
i can see nothing in the code that would prevent that from using the "spi" bus to the mixer while sweep_mode() is also accessing the "spi" bus for retuning.

so the solution would be either some way to drop out of sweep_mode() that doesnt involve mixer spi.
or some kind of locking that prevents the concurrent access.

gozu42, 2021-03-13:
@miek asked me to add the details of how i got to "its the mixer" and "why hackr_close".
the following is a summary of my notes and rants about the process on various chats, so even
if something there looks like a question, it may just have been a question at the time, and resolved by now.

my captures are mostly hackrf_sweep at 100kHz binsize for full 0-6Ghz span, which reports 1.35 sweep/s.
the aggregation in the images is 60sec vertical and 4MHz horizontal per pixel.
the scaling is full 0-255 value range per channel first, changes to 0-127 on red later.
(since min seems to be the least interesting channel and the glowing red was annoying me)

i was using the release version (early 2018) first, but switched to git head 2021-03 later. no difference.
the antenna is a ca 50cm telecoping monpole that came with the kit, which explains the "mostly deaf in 3-6GHz".

this is a pic where the acquisition method was to run hackrf_sweep processes for 10 minutes each, then kill it:
https://i.imgur.com/QP7oq9l.png

this looks similar to problem (A) in the original post, with the added twist of the stepped/quantized fadein/out.

i have some that may be a variant of problem (B) too, like the higher end of the low band and high band ghosting here:
https://i.imgur.com/DEq5mMi.png

and this is a pic where hackrf_sweep was set to -N 1000 (== 1000 sweeps) and the scheduler set to wait 2min before restarting.
this means the black lines (== no data) represent "restarts of hackrf_sweep").
https://i.imgur.com/RtWZb2T.png

this makes it pretty clear that "good" and "bad" depend on the hackrf_sweep process running.

all these also make it clear that this is somehow related to the band scanned:
the low+high bands "vanish" entirely, the middle band (== bypass) is fine.
after reading some firmware code, this seems to indicate some problem with either "mixer" or "switchcntrl".

going by https://www.dolstra.nl/Ham-radio/SDR_Tranceivers/HackRF%20One%20SDR/hackrf1-frontend-20140213.png
just the switch directly by RX IF permanently stuck in BYPASS (up) mode would explain it.
otoh, the mixer playing dead would also explain this.
neither (to me) explain why the signal in the low band is "stepped" in the first pic.
(note how it goes quantized weaker before going fully dead, and reverse when returning)
or the "ghosting" in the B case. perhaps the pass-filter switches stuck the wrong way?
otoh, i dont understand the mixer enough, and it being in a _weird_ configuration might explain all of this.

my workaround has been a bit more "software" than "arduino powercycling the hrf":
i added a hackrf_reset+hackrf_close+sleep+hackrf_open sequence to hackrf_sweep after the original hackrf_open.
the hostside USB debug messages indicate it is indeed resetting, and the lack of such messages over
the lack of such msgs over the last week indicates the "mysterious recoveries" were not the hrf resetting on its own.

so the problem is some kind of state cleared by "hackrf booting" one-time-init.
but not by hostside hackrf_sweep startup one-time-init.

since the problem is not just "during the first sweep", it indicates some sticky problem.
since "it seems to randomly recover after some runs", my guess is more "hackrf_close" than "hackrf_open".
like "what if it shuts down in the middle of an spi transaction with the mixer".

to narrow it down more, i added a mixer_resetup() to the firmware that pulls the mixers gpio_reset low,
waits for 1mil nops, then calls mixer_setup() again.
added a call to this (and mixer_rx()) at the beginning of usb_api sweep_mode().

this is not meant as an actual fix (so i wont share the code unless someone _really_ wants it), but
since that "fixes" the issue, it is clearly the mixer being bad.

it being related to "the end" of the hackrf_sweep process is more likely than "the start" since i have never seen it
happen on a freshly restarted hackrf.
from that it was a bit of cold-reading the shutdown code, and looking for anything that touches the mixer.

and hackrf_close() calls set_transceiver_mode(OFF) which calls mixer_disable() ...

mossmann, 2021-03-17:
Thanks a lot for all the details, @blardyblar and @gozu42, and for the fix, @miek!

closed 2021-03-17 by mossmann

gozu42, 2021-03-18:
ah, i am too slow, but fwiw ... 

i have been running mieks fix for ~30h now with a restart of the hackrf_sweep process every 1000 sweeps (~12 minutes), and it the problem didnt appear.

so works for me, and thanks to everyone involved!

PR #847 (2021-03-17): Disable USB interrupts during sweep set_freq call
Previously, a USB vendor request could interrupt in the middle of
`set_freq` and make conflicting changes, leaving the HackRF in an
undefined state.

fixes #772
commit 1cb2abf7 (2021-03-17): Disable USB interrupts during sweep set_freq call

Previously, a USB vendor request could interrupt in the middle of
set_freq and make conflicting changes, leaving the HackRF in an
undefined state.

fixes #772