greatscottgadgets/hackrf #774: HackRF + Portapack 2 problem upper than 2700 MHz

reclaudio, 2020-08-10:
Thanks a lot for this marvelous work !
I have recently bought an HackRF + Portapack 2 with v1.1.0 installed .
I have done some test on the Signal Generator section and looks very fine considering the costs involved .
I have noticed some problems :
A) The output frequency doesn't go upper than 2700 MHz . You can write it , but can't exceed 2700 MHz , after it doesn't generate . Rotating the knob you can also exceed 2700 MHz , but arriving at 2700 MHz it really stops there .
B) I wasn't able to understand how and if the BW setting change something , the modulation looks always the same
C) I wasn't able to generate a pure carrier , if not using the Noise modulation that seems not to work , or using a Sine modulation with frequency = 0
D) I didn't understand how to use the CW modulation . I tried to push the buttons ....

Is where something that I am doing wrong or some problems was corrected in the last firmware version ?

Of course the bigger limitation is point A .

I was told from the Github of Mayem that I have a defective Hardware HackRF .

Anyone had the same experience and can suggest solutions . I don't have big problem in try to fixing even complicated boards .

Thank you and kind regards .

Claudio Re I1RFQ

miek, 2020-09-10:
Are you able to reproduce the problem in HackRF mode, connected to a PC & and using something like GNURadio or similar to generate the signals?

straithe, 2021-01-26:
Are you still having trouble? We will close this issue if we don't get a reply within the next month.

reclaudio, 2021-01-26:
Hello ,

          thank you for your message .

I am glad that you asked me .

 

I stopped to claim because so far I always got the same answer : the software is fine , your hardware is bad (no specification if on the mainboard or on the HackRF) .

 

For curiosity I made further measurements in August 2019 of Portapack 2 and HackRF with two spectrum Analyzers ( HP8560E and 8563E) and also driving directly the HackRF of Portapack 2 with SDR Angel .

 

Of course I can find in my record pictures and video I have made and share with you all this technical measurements and collaborate .

 

I could be wrong , but if I remember well there was a frequency hole on all app at 2700 – 2758 MHz with Portapack2 and his HackRF , but strangely the hole was only from 2750 to 2758 MHz driving  with SDR Angel .

 

Also I tried to record and playback signals and there was problems of aliasing .

 

I made a video that show this issue .  

 

I think we meet in Geneva at  Radio Hack at ITU .

 

Please let me know if you want to have the technical data I have measured , and if yes here or privately .

 

BTW , I also have here other two HackRF Great Scott Gadgest without Portapack that could help in debugging the problem , if hardware on the HackRF. 

 

Thank you and kind regards .

 

Claudio Re 

A: mossmann/hackrf
Cc: reclaudio; Author

 

Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

reclaudio, 2021-01-27:
Hello , 

           just a note .

The claims that the problems depends on the hardware and not from the software comes from the Github list portapack-mayhem .

Please let me know if you are interested to have what I have measured with the  HP spectrum analyzers driving the hardware with the mayem software and with SDR Angel .

73 – Claudio

 

Da: Claudio Re [mailto:reclaudio@alma.it] 
Inviato: martedì 26 gennaio 2021 16:33
A: 'mossmann/hackrf'; 'mossmann/hackrf'
Cc: 'Author'
Oggetto: R: [mossmann/hackrf] HackRF + Portapack 2 problem upper than 2700 MHz (#774)
Priorità: Alta

 

Hello ,

          thank you for your message .

I am glad that you asked me .

 

I stopped to claim because so far I always got the same answer : the software is fine , your hardware is bad (no specification if on the mainboard or on the HackRF) .

 

For curiosity I made further measurements in August 2019 of Portapack 2 and HackRF with two spectrum Analyzers ( HP8560E and 8563E) and also driving directly the HackRF of Portapack 2 with SDR Angel .

 

Of course I can find in my record pictures and video I have made and share with you all this technical measurements and collaborate .

 

I could be wrong , but if I remember well there was a frequency hole on all app at 2700 – 2758 MHz with Portapack2 and his HackRF , but strangely the hole was only from 2750 to 2758 MHz driving  with SDR Angel .

 

Also I tried to record and playback signals and there was problems of aliasing .

 

I made a video that show this issue .  

 

I think we meet in Geneva at  Radio Hack at ITU .

 

Please let me know if you want to have the technical data I have measured , and if yes here or privately .

 

BTW , I also have here other two HackRF Great Scott Gadgest without Portapack that could help in debugging the problem , if hardware on the HackRF. 

 

Thank you and kind regards .

 

Claudio Re 

 

https://www.avast.com/antivirus

mossmann, 2021-02-04:
> I could be wrong , but if I remember well there was a frequency hole on all app at 2700 – 2758 MHz with Portapack2 and his HackRF , but strangely the hole was only from 2750 to 2758 MHz driving with SDR Angel .

The HackRF firmware configures a different RF path for frequencies above 2750 MHz than it does for frequencies below 2750 MHz. Alternative firmware such as mayhem might have a different boundary. It is possible that the hardware may be having difficulty tuning to frequencies on one side of the boundary, and it is also possible that a modification to the firmware could be made that avoids that problem.

> I made a video that show this issue .

If you attached a video to an email reply, it was stripped by GitHub. Could you please visit the issue in a web browser to attach or link to your video?

reclaudio, 2021-02-04:
Hello ,

           at the following link you can download a medley of the pictures I have made in the past in sweeping both HackRF in his box and a Portapack 2 with HackRF manually and automatically , or setting various frequency and the video I have with comments in English .

If you need further info or measurements I will be happy to perform them and send you .

https://www.dropbox.com/s/2yvu2bn4x2zd0k3/Portapack%202%20%26%20HackRF.rar?dl=0

 

73-Claudio I1RFQ  

 

https://www.avast.com/antivirus

mossmann, 2021-02-20:
Thank you for the photos and video. I don't understand what some of them represent, but I can see that your HackRF One is producing output at 2700 MHz but not at 2701 MHz.

Can you please try installing firmware from the 2018.01.1 release (or from git master if you are comfortable compiling it yourself)?

Then try these two commands to see if they are able to produce output at 2701 MHz:
`hackrf_transfer -a 0 -x 47 -c 127 -f 2701000000`
`hackrf_transfer -a 0 -x 47 -c 127 -i 2200000000 -o 501000000 -m 2`

While using the regular HackRF firmware, you will lack Portapack functions.

straithe, 2021-03-29:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-03-29 by straithe

mossmann, 2022-12-09:
This may be a symptom of the bug fixed in #1237.