greatscottgadgets/hackrf #775: No Data Transferred error when running hackrf_sweep as a python subprocess

dails08, 2020-08-13:
### Steps to reproduce
1. In python 3.8, run code:
`proc = subprocess.Popen(["hackrf_sweep"], stdout = subprocess.PIPE, text=True)`

### Expected behaviour
I'd expect the stdout to be sent to the stdout pipe of the subprocess.  hackrf_sweep seems to receive no data.  When using the same code, but sending the data to /dev/null:
`proc = subprocess.Popen(["hackrf_sweep"], stdout = subprocess.DEVNULL, text=True)`
it works fine.  This behavior is the same on a RPi4 and an ubuntu laptop.

### Actual behaviour
The stderr output is:
```
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 6000 MHz
Stop with Ctrl-C
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second

Couldn't transfer any data for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 0 in 0.50112 seconds (0.00 sweeps/second)

```

### Version information
**Operating system**:
Ubuntu 20.04 and Raspberry Pi Os 10 (buster)

**hackrf_info output:**
```
hackrf_info version: git-6e5cbda
libhackrf version: git-6e5cbda (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000057b068dc238ea363
Board ID Number: 2 (HackRF One)
Firmware Version: git-6e5cbda (API:1.04)
Part ID Number: 0xa000cb3c 0x0058435a
```

miek, 2020-08-20:
It's timing out on data transfer because nothing is reading from the stdout pipe. It works for me if I run this instead:

```
proc = subprocess.Popen(["hackrf_sweep"], stdout = subprocess.PIPE, text=True)
while True:
    print(proc.stdout.readline())
```

dails08, 2020-08-20:
That makes sense! I'm surprised to learn that it checks that; I'd have thought it wouldn't be able to know it was writing into the buffer. Maybe there's a set size for the pipe buffer and the data fills it very quickly. In any case, that's enough insight for me to call it closed. Thanks!

closed 2020-08-20 by miek