greatscottgadgets/hackrf #776: Shorted burned up an ic on the board using an amplifier

alexxmarsh44, 2020-08-18:
I plugged an LNA amplifier into the antenna port it and it immedietly smoked and i think I identified the ic that burned up. 

[image]

Which part is this?  I want to attempt to replace it.  Do you think its possible that it is an isolated part that burned up or are their other likely ones that went with it?

cjheath, 2020-09-06:
This is the MOSFET switch for the bias T that feeds power to the external LNA.
You will damage it if the antenna port is short circuit or draws too much power.

miek, 2020-09-10:
I think this could also be caused by the LNA feeding back a high enough bias voltage on to the antenna port, if it was an externally powered LNA?
It's hard to tell if that's the only failed part, a good test would be to de-solder it and then check functionality.

cjheath, 2020-09-10:
If you don't use the Bias T, the whole HackRF performance can be improved by removing the inductor (just bottom-left of the red square). It loads the antenna port unnecessarily if you don't need it. I tried to improve the Bias T by adding a second stage (smaller inductor) but don't yet have measurements on how successful it was.

alexxmarsh44, 2020-09-18:
I now got it working temporarily as it turns on again.  As soon as I use it in gnuradio to receive it starts blinking the 3v3 1v8 rf and usb lights continuously at the same time.  It gives a shorted error in gnuradio at the same time.  It only works for maybe twenty seconds.  If I attempt to use it for something less strenuous it will start flashing but before it does I noticed the board getting very hot much hotter than usual.  Any ideas on how to fix it?  Ill try anything before chocking it up to a loss.

cjheath, 2020-09-18:
@alexxmarsh44 Even if you can't replace the MOSFET, you can remove it with wire clippers. This should restore normal function of the HackRF, but you won't be able to power an external LNA.

mossmann, 2021-01-24:
Are you still looking for help with this problem? We will close the issue if we don't see a reply within the next month.

closed 2021-02-23 by straithe