greatscottgadgets/hackrf #777: How to print coming data with HackRFone

sametozgul, 2020-08-23:
Hello Everyone, 
I'm trying to print data when HackRF is RX mode. But I don't understand how to do. I just changed rx_callback() function like below.

int rx_callback(hackrf_transfer* transfer) {
	size_t bytes_to_write;
	size_t bytes_written;
	int8_t* buf;
	unsigned int i;
        bytes_to_write=transfer->valid_length;
	buf = (int8_t*) transfer->buffer;
	while(a<bytes_to_write){
		fprintf(stderr,"%d\n ", buf[a++]);
	}
	
	
}

And result like this:

 -2
 1
 -3
 2
 -2
 1
 -2
 1
 -2
 1
 -2
 1
 -2
 1
 -2
 1
 -2
 1
 -2
 1
 -2
 1
 -2
 2
 -2
 1
 -2
 1
 -2
 1
 -2
 1
 -2
 1
 
I would like to read IQ data and detect coming signal from other HackRF. I don't want to write or read file. Can someone help me ?

miek, 2020-08-23:
That looks correct to me. The first byte you print is I, the next is Q, then I, Q, I, Q, etc. They're all hovering around 0, so the signal level is probably very low.

sametozgul, 2020-08-23:
> That looks correct to me. The first byte you print is I, the next is Q, then I, Q, I, Q, etc. They're all hovering around 0, so the signal level is probably very low.

I think, The Signal is enough. Because when I try to read same RF signal with ADALMPluto I can detect all peaks. How can I detect peaks in this result ?

miek, 2020-08-23:
I recommend testing with the HackRF itself, using an existing tool like GQRX/GNURadio/etc., so you can figure out the correct gain settings first.

sametozgul, 2020-08-24:
I changed the gain as in GNURadio but nothing changed. When I use GNURadio , I see what I expect. ( This source code is actually hackrf_transfer.c ,I just changed rxcallback function to print data) What should I do ? Does anyone has "receive_example.c etc."?

closed 2021-07-18 by straithe