greatscottgadgets/hackrf #780: Is it worth it to build it myself to save money or it'll cost more?

noviceFedora, 2020-08-26:
Presently this is being sold for around $400 in India. How much can I save by building it myself?

Where can I find a list of minor parts and how many of each of them I should buy. Are these parts sold in bulk or can they be bought in quantities required by this project?

zjywlive, 2020-08-30:
> 目前，该产品在印度的售价约为400美元。我自己建造可以节省多少？
> 
> 在哪里可以找到次要零件的清单，以及应该购买的零件数量。这些零件是批量出售还是可以按项目要求的数量购买？

Hackrf is now available in China for less than $150, and if you want to add a portapack, you can pay less than $60

zjywlive, 2020-08-30:
> Presently this is being sold for around $400 in India. How much can I save by building it myself?
> 
> Where can I find a list of minor parts and how many of each of them I should buy. Are these parts sold in bulk or can they be bought in quantities required by this project?

Hackrf is now available in China for less than $150, and if you want to add a portapack, you can pay less than $60

noviceFedora, 2020-08-31:
> Hackrf is now available in China for less than $150, and if you want to add a portapack, you can pay less than $60

Do you mean to say if I buy HackRF with Portapack, I have to pay less than $60?

With the current pandemic, it will be a good idea to import from China. I rather want to build it to gain experience.

eried, 2020-08-31:
I think you should assemble and document the process and BOM prices, etc. It will be a nice soldering learning experience.

noviceFedora, 2020-08-31:
> I think you should assemble and document the process and BOM prices, etc. It will be a nice soldering learning experience.

Where is the list of minor components? Do you know if they are sold in the quantities required by the project or more?

zjywlive, 2020-08-31:
> > I think you should assemble and document the process and BOM prices, etc. It will be a nice soldering learning experience.
> 
> Where is the list of minor components? Do you know if they are sold in the quantities required by the project or more?

The list can be generated by KiCAD

zjywlive, 2020-08-31:
> > I think you should assemble and document the process and BOM prices, etc. It will be a nice soldering learning experience.
> 
> Where is the list of minor components? Do you know if they are sold in the quantities required by the project or more?

In China, some components can be ordered in any quantity (because they are more expensive), while others cannot, such as small capacitors or resistors (they are very cheap)

miek, 2020-09-10:
Generally it's not worth self-building to save money, but it can be a very nice project to learn from. However, I wouldn't recommend it as a first (or second or third..) SMD soldering project. There's a lot of placements and fine-pitch parts, so you should get some experience building simpler projects first if you haven't already.

closed 2020-09-10 by miek