greatscottgadgets/hackrf #781: Is HackRF One good for sweeping for bugs and implants?

noviceFedora, 2020-08-31:
Is HackRF One good for sweeping for bugs and implants?

I have a DVB-T stick, I'm running it continuously with rtl_power, sweeping it's entire range, I'm seeing the usual services, FM radio stations, and many other spikes which I'm not able to make sense of, it could be possible there could be some bugs and implants which are operating at frequencies higher or lower than the frequencies covered by the DVB-T stick I have. 

Does anyone here know what frequencies bugs and implants typically operate on? Is HackRF One suitable for this job for someone? How can I sweep for bugs and implants?

eried, 2020-08-31:
HackRF is not very sensitive. Compared to my cheap chinese sdr tv stick is just slightly better on some ranges. I would opt for the HackRF because it goes from 10 mhz to 6 ghz, not because its sensibility...

noviceFedora, 2020-08-31:
> HackRF is not very sensitive. Compared to my cheap chinese sdr tv stick is just slightly better on some ranges. I would opt for the HackRF because it goes from 10 mhz to 6 ghz, not because its sensibility...

Okay, I know it's range goes up to 6GHz. Which specifications show it's sensitivity? What sensitivity should I be look for sweeping bugs? Shouldn't it be enough to capture bugs and implant activity if it can capture satellite signals, etc?

Javantea, 2020-11-12:
I was able to use the HackRF One to find a noisy bluetooth mouse in my office. It's pretty good at finding signals nearby to it. My Alinco DJ-G7 has a bug-finding mode which checks a set of common bug frequencies which you can probably imagine isn't very effective -- checking 100 frequencies doesn't provide peace of mind. With HackRF, you can find tons of signals with the frequency plot or waterfall plot, and if they are nearby you can locate them. Your task is difficult, though, so I don't think anyone will give you a guarantee that HackRF can find bugs.

closed 2021-01-23 by mossmann