greatscottgadgets/hackrf #782: libhackrf device discovery issue

rgetz, 2020-09-03:
It was recently reported that libiio was causing libhackrf not to discover hackrf devices (which not on purpose, and needs to get fixed).

Original issue:
https://github.com/pothosware/PothosSDR/issues/69

it was suggested (but not confirmed) it might be was because the way that libiio (which is a generic interface for any IIO device, not just PlutoSDR - so it doesn't include PID/VID) walks through the all available USB devices, looking for the one with "IIO" as a device attribute string. It appears that libusb does not immediately allow other threads/processes to look at these, and just returns errors when something attempts to open it. (from my testing - it seems to take a ~10 ms or so to release the handle back to the system).
https://github.com/analogdevicesinc/libiio/blob/master/usb.c#L486-L513

I tried replicating the same with `while[1] ; do sudo lsusb -vvv >/dev/null 2>&1` ; done` (which should be doing something similar to libiio - just walking through some usb device descriptors) while doing a search for usb based iio devices - and couldn't cause a failure in ~5 min and was wondering if this was something specific to libhackrf? (I left running overnight to see if it was just really random).

Thanks
-Robin

straithe, 2021-11-12:
Hi, @rgetz. Were you able to confirm the issue? Is it still occurring?

rgetz, 2021-11-12:
I don't have a hackRF to test with (sorry).

straithe, 2021-11-12:
Thank you for the update. I am really sorry that we did not get to this question while it was still fresh. 

I haven't seen any other HackRF GitHub issues opened where a person was experiencing the same thing. It also seems a lot of the software you mention has been updated since your original message positing. Since there aren't reproducibility instructions and you don't have the same setup as before, I'm going to close this issue. If anyone else does experience this I'll be happy to re-open this issue or get another one started.

rgetz, 2021-11-12:
No problem - thanks for following up - I don't think we did anything to libiio to "fix" / "change" this yet (still on the list). If you want to test, just do a :
```
while[1] ; do sudo lsusb -vvv >/dev/null 2>&1
```
at the same time you are looking for hackrf on USB. I assume that hackrf_device_list_open() does some sort of usb scanning?

I can also provide a pluto or M2k to add to your test bench - let me know and I can send any over - and if no  - that's understandable too.

straithe, 2021-11-12:
This helps a little, but I still have some questions like: 
What OS is this on? 
Is there a VM involved? 
What version of the libhackrf? 
What version of the hackrf firmware?

I don't have a pluto or M2k personally. You've made a very generous offer. I'll check if the GSG lab has one so I can do some extra testing without having to cost you hardware. :)

rgetz, 2021-11-12:
I don't know if you can get any version info from the original ~ Jun 2019  bug report:
https://github.com/pothosware/PothosSDR/issues/69

Looks like Windows, but was awhile ago.

Feel free to DM me on twitter : robinlgetz with a shipping address - I don't mind supporting GSG, and it's always nice to see what others are doing.

straithe, 2021-11-12:
Following the rabbithole of issues it looks like people using RTL-SDR are experiencing the same initial issue you were talking about, so I don't think it is libhackrf. I think https://github.com/pothosware/SoapyPlutoSDR/issues/24 adds some more context and information. :)

gozu42, 2021-11-12:
(speaking from the sideline here. i am not a GSG person, i just like solving riddles. and oh wow, this one is a quality complex of rabbitholes...)

tbh, if i have 4 kinds of SDR device connected to the same host and running 3 different applications against them, i do not expect things to work out 100% smooth all the time, which sometimes has me cursing, but i also spend more time trying to figure out in what order to start things instead of filing tickets about it.

my favorite summary of this issue is here: https://github.com/analogdevicesinc/libiio/issues/586
i absolutely agree with the "I dont think I can convince HackRF, RTLSDR, LimeSuite, USRP, and potentially un-enumerable others to put retry loops.".
which pretty much closes this issue here on the GSG side.

otoh, it still is an issue that seems worth solving, and for that, the much more recent issue here seems like a good vector: https://github.com/pothosware/SoapyPlutoSDR/pull/42
not sure why this isnt merged.
@rgetz do you have any additional vid/pid to add there, be they official or known less than official?
@guruofquality @zuckschwerdt is there any reason besides "waiting for more vid/pid" to not merge that? (if not, please just merge it, adding more IDs as they pop up seems like a really trivial thing to do.)

@straithe @rgetz you two should still absolutely swap a pluto for a hackrf, for science.

closed 2021-11-12 by straithe

zuckschwerdt, 2021-11-12:
The patch pothosware/SoapyPlutoSDR#42 is a check to see if an already open IIO context has all the devices we expect, it won't help with enumeration.
My Idea outlined there was to abort before any IIO enumeration if there is no known Pluto VID:PID, and otherwise pause (how long? no idea. 50, 500 µs, longer?) so other drivers get a chance first. I never encountered the race condition, we need somebody to reliably reproduce the race and test fixes.

straithe, 2021-11-12:
Thank you for the additional information.

zuckschwerdt, 2021-11-13:
To everyone experiencing this usb enumeration problem: please try https://github.com/pothosware/SoapyPlutoSDR/tree/fix-race and experiment with values for the wait in https://github.com/pothosware/SoapyPlutoSDR/blob/fix-race/PlutoSDR_Registration.cpp#L19 -- does this help? What is the lowest wait that reliably works?

zuckschwerdt, 2021-11-14:
Also https://github.com/pothosware/SoapyPlutoSDR/tree/feat-libusb aborts the usb enumeration if not at least one ADALM-Pluto VID:PID (0456:b673) is found.